radioconda/radioconda-installer #1: radio beginner question

bbartling, 2021-04-06:
Hi Ryan,

I am new to the world of radio communication and I apologize for a git issue to ask these questions. But this package seems cool I am familiar with Anaconda for data analysis purposes.

Most of my work in experience has been setting up building automation systems in the field and I am recently working in more an IoT role for demand response, smart grid applications where a utility provider can send a signal to the building to trim power via open automated demand response ([open ADR](https://www.openadr.org/)) which is mostly just an XML based HTTP request standard. In the open ADR community webinars there is always talk on using FM radio wave to communicate demand response events and I have been researching more about how smart utility meters already use FM broadcast signals/mesh networks, [this method](http://www.arrl.org/smart-meters). And I know the utility industry already has been using FM signals I think for decades now..

The question I have could I ever use this package to setup an experiment, like a basement hobby project to transmit a message from one computer to another via FM radio/GNU radio? For example could a signal be generated on a "server" computer "demand response level" which can be either a "0,1,2,3" and "event duration in hours"  be sent from one computer to another via FM radio?

I'm at the beginner level of taking some Python based signal processing refresher courses as well. Its been a solid 10+ years since my college course work that briefly covered this.

Any tips or roads to go down greatly appreciated for .

Thanks,
Ben

ryanvolz, 2021-04-06:
Hi Ben! This certainly sounds like something one could do with GNU Radio with some relatively cheap hardware. People regularly use it to test various modulation and encoding methods, including FM, between a transmitter and receiver, although the rabbit hole is deep once you start with communications protocols. I'm a radio science researcher myself, so I mostly know of the communications applications through observation and not much experience.

Within the GNU Radio community, there is the main "GNU Radio" software package that is included here that includes the framework and some common processing blocks. Then there are a bunch of user-created out-of-tree (OOT) modules that provide processing blocks for various more specific applications. A quick search leads to https://github.com/argilo/gr-elster for decoding some types of smart meters, but there might be more out there. So if you want to try this out, you could install GNU Radio (e.g. with radioconda), get some radio hardware that can transmit and receive in the frequency bands you need for the smart meters (ADALM-PLUTO might be a good start), and then play around with it using the included blocks or by installing some OOT modules (https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio). GNU Radio has a [mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) and also a [Matrix chat server](https://chat.gnuradio.org/) where people with more experience in your application can give you more pointed directions.

I hope that's helpful!

bbartling, 2021-04-06:
Thanks for the info going to install radioconda soon here... Do you recommend Linux, or is Windows Ok?

ryanvolz, 2021-04-06:
I definitely recommend Linux, as most of these tools are primarily developed on Linux. However, if it's Linux in a VM or Windows Subsystems for Linux versus native Windows, then you should probably try native Windows first. Just know that while I've done my best to test what things I can on Windows, there are not nearly as many Windows users for these software packages in general and you're more likely to run into issues just because nobody else has tried to do the thing you're trying. But I heartily welcome Windows users who report bugs so that these things can be improved. :smile:

bbartling, 2021-04-07:
Hi Ryan,

For a basement hobby project sending a radio signal from one device to another would these two products be compatible?

AdaFruit radio transmitter - $20

AdaFruit SDR radio receiver - $22

I the radio transmitter appears to use a microphone input but I was curious if radioconda could be used instead? As well as radioconda on the receiver end to decode the GNU radio message?

Thanks for anytime you have in response.
Ben

ryanvolz, 2021-04-07:
> [AdaFruit radio transmitter](https://www.adafruit.com/product/1958?gclid=Cj0KCQjwsLWDBhCmARIsAPSL3_3Q5uJch1vtf-U00BbqcpNDQ5oYb9ULk18Q3BJBC5SYGdvvogKexewaAvEqEALw_wcB) - $20

This looks like just a hardware audio->FM transmitter, so that's probably not something you want as it is very limited in what it can do. I doubt you could make any signal that acts like the smart meters, for example.

> [AdaFruit SDR radio receiver](https://www.adafruit.com/product/1497) - $22

That is an RTL-SDR device, so that is a generic software-defined-radio receiver and it would work with the RTL-SDR drivers that are packaged with radioconda. That said, [one of these](https://amzn.to/2nIKFcT) is what I have and I would recommend it if it were in stock. It has some upgrades including a better connector that would let you hook up different antennas in the future if necessary. I've also heard good things about [this one](https://www.nooelec.com/store/sdr/nesdr-smart.html) although it is a bit more expensive. [This blog post](https://blog.bliley.com/10-popular-software-defined-radios-sdr) goes over a few other radio options (not all of which currently have drivers in radioconda, but they could be supported eventually).

The transmitter is where you really run into difficulty at low cost. The nicest, cheapest product I'm aware of is the [ADALM-PLUTO](https://www.analog.com/en/design-center/evaluation-hardware-and-software/evaluation-boards-kits/adalm-pluto.html). The next step up is the [HackRF One](https://greatscottgadgets.com/hackrf/one/) (not supported in radioconda, yet) or the [LimeSDR or LimeSDR mini](https://limemicro.com/products/boards/). Higher end you're talking one of the [Ettus USRPs](https://www.ettus.com/products/). None of those are very cheap, but they are all quite capable. If you come across anything else that looks interesting, let me know and I'll see what supporting it might require. Given the expense of these and the cheapness of the AdaFruit transmitter, it might make sense to just get one of those anyway for playing around (I don't think you'd need any special software, just the ability to play sound into an audio jack).

bbartling, 2021-04-07:
Ryan,

Is this [yard stick device](https://www.adafruit.com/product/3586) also just another non-compatible radio transmitter? Non-compatible with GNU radio/radioconda, etc...

I noticed it mentioned transmit and receive.

ryanvolz, 2021-04-07:
It doesn't look like it uses GNU Radio or anything packaged with radioconda, but it also looks like it does its own thing in firmware that you can control from your host computer. So it might do what you need, maybe? At least it claims to, but I don't know how it works exactly.

ryanvolz, 2021-04-07:
Check the frequency ranges on that though and make sure they cover what you want. It's a bit more restrictive than the more expensive devices.

bbartling, 2021-04-07:
Thanks for all the help Ryan, going to close this for now :)

closed 2021-04-07 by bbartling

bbartling, 2021-04-11:
Hi Ryan,

Curious to bug you about one more question if you have the time I ran across [this article ](https://medium.com/dev-genius/diy-air-transfer-transmit-and-receive-data-using-gnu-radio-and-a-microphone-f0adc8f1f563)on Medium for a DIY air drop from one device to another using GNU radio and a microphone.

Could I ever use `radioconda` for the send and receive? The author of the medium story mentioned about using UDP on the receiving end which I still have to wrap my head around what he is doing with sockets, etc... But this post seems like a fun experiment for an absolute beginner. Can you run the GNU blocks from `radioconda` directly? Would you have any examples how to do that by chance with `radioconda`?

I still have to learn a ton about when radio transmission on the level of computer speakers with a microphone Versus radio transmission with a transmitter device (like Hack RF One) and an antenna where at that point I think probably look into getting a HAM license.

As always thanks for anytime you would have in response.
Ben

ryanvolz, 2021-04-11:
Yes, you could do that. All you need is GNU Radio to create and run the flowgraphs. The UDP sockets are just one way to be able to send data to another program, and in this case it's just a simple Python script. I'm sure the same thing could have been done directly in GNU Radio, maybe even with some built-in blocks but definitely with an embedded Python block.