radioconda/radioconda-installer #2: Upgr

MountainLogic, 2021-04-19:
I did have radiconda working uder win10, but I just did a mamba upgrade all and now when I go to start/Gnu Radio Companion (Radioconda) App, I have a common box pop-up ever so briefly, then it goes away.  The Radioconda command line has no path to find gnuradaio-companion.  Not sure what my next stops would be.   BTW, also looking to use IIO on win10.

ryanvolz, 2021-04-19:
That may have upgraded you to GR 3.9, which could be the issue since it hasn't had much end user testing, especially on Windows.

Can you open a Conda prompt and post the output of `mamba list`? That will tell us what is still actually installed. It's curious that you can't even run GRC from the command line, which is something I would normally recommend for figuring out weird launch issues.

ryanvolz, 2021-04-19:
Oh and I forgot to respond to the IIO part. `iio` should work as far as I know, but do you mean `gr-iio` for the GNU Radio blocks? For that, I haven't been able to add it to conda-forge because upstream hasn't made an official release that was compatible with GR 3.8 or 3.9. As you may know though, it's in the process of being included into the main GR repository, so I'm hopeful to be able to have it available for GR 3.9 before too long.

MountainLogic, 2021-04-19:
I saw some promises about gr-iio being included in 3.9, oh well, Analog Devices....

BTW, I'm usually also on gr chat.

`
mamba list                                                                                                                                                                                                                               __    __    __    __                                                                                                   /  \  /  \  /  \  /  \                                                                                                 /    \/    \/    \/    \                                                                                ███████████████/  /██/  /██/  /██/  /████████████████████████                                                                         /  / \   / \   / \   / \  \____                                                                                        /  /   \_/   \_/   \_/   \    o \__,                                                                                   / _/                       \_____/  `                                                                                   |/                                                                                                                  ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗                                                                           ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗                                                                          ██╔████╔██║███████║██╔████╔██║██████╔╝███████║                                                                          ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║                                                                          ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║                                                                          ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝                                                                                                                                                                                                  mamba (0.10.0) supported by @QuantStack                                                                                                                                                                                                         GitHub:  https://github.com/mamba-org/mamba                                                                             Twitter: https://twitter.com/QuantStack                                                                                                                                                                                                 █████████████████████████████████████████████████████████████                                                                                                                                                                                   # packages in environment at C:\Users\selliott\radioconda:                                                              #                                                                                                                       # Name                    Version                   Build  Channel                                                      adwaita-icon-theme        3.38.0               heaa5689_1    conda-forge                                                appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge                                                argh                      0.26.2                py38_1001    conda-forge                                                atk-1.0                   2.36.0               h7222f49_4    conda-forge                                                backcall                  0.2.0              pyh9f0ad1d_0    conda-forge                                                backports                 1.0                        py_2    conda-forge                                                backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge                                            boost-cpp                 1.74.0               h54f0996_2    conda-forge                                                brotlipy                  0.7.0           py38hab1e662_1001    conda-forge                                              bzip2                     1.0.8                h8ffe710_4    conda-forge                                                ca-certificates           2020.12.5            h5b45459_0    conda-forge                                                cached-property           1.5.2                hd8ed1ab_1    conda-forge                                                cached_property           1.5.2              pyha770c72_1    conda-forge                                                cairo                     1.16.0            hb19e0ff_1008    conda-forge                                                certifi                   2020.12.5        py38haa244fe_1    conda-forge                                                cffi                      1.14.5           py38hd8c33c5_0    conda-forge                                                chardet                   4.0.0            py38haa244fe_1    conda-forge                                                click                     7.1.2              pyh9f0ad1d_0    conda-forge                                                cli [... cut ...] forge                                                yaml                      0.2.5                he774522_0    conda-forge                                                zeromq                    4.3.4                h0e60522_0    conda-forge                                                zlib                      1.2.11            h62dcd97_1010    conda-forge                                                zstd                      1.4.9                h6255e5f_0    conda-forge            

`

ryanvolz, 2021-04-19:
It looks like everything is there. Can you see if `gnuradio-config-info` will run or turn up any issues? Can you see if `%PREFIX%\Library\bin\gnuradio-companion.exe` exists and if you can run it from there with the explicit path? (Other notable executables should also be in the same directory.)

MountainLogic, 2021-04-20:
Was a conflict with pathos.  Once pathos was removed all seems to be better.  Think you can close this.  Details from radioconda install:

Interesting, no gnuradio-config-info seems to be preset in the path ( gnur then tab fails to show me any completion options).

```
(base) C:\Users\foobar\radioconda\Scripts>gnuradio-config-info --prefix
C:\Program Files\PothosSDR
 
```

So that reminds me that I did  install pathos, so that may be the problem. 

Interesting that gnuradio-companion.exe is not in the radoconda script directory.  Shuold three not be a few more GnuRadio bins in here? 

Inside radioconda command line I ran:
```
                                                                                                                        
Directory of C:\Users\foobar\radioconda\Scripts                                                                                                                                                                                              11/01/2019  02:30 PM            40,960 2to3.exe 
10/26/2020  08:42 AM            41,984 chardetect.exe
04/14/2021  09:40 AM           74,752 conda-env.exe                                                                    
04/14/2021  04:15 PM            74,752 conda.exe                                                                        
01/15/2021  03:04 AM            41,984 cph.exe                                                                          
02/12/2021  05:47 PM            74,752 drf.exe                                                                          
10/26/2020  08:42 AM            41,984 easy_install.exe                                                                 
01/15/2021  03:04 AM            41,984 f2py.exe                                                                        
01/15/2021  03:04 AM            41,984 gnuradio-companion.exe                                                           
01/15/2021  03:04 AM            41,984 grcc.exe                                                                         
01/15/2021  03:04 AM            41,984 gr_filter_design.exe                                                             
01/15/2021  03:04 AM            41,984 gr_modtool.exe                                                                   
11/01/2019  02:30 PM            40,960 idle.exe                                                                        
01/15/2021  03:04 AM            41,984 iptest.exe                                                                       
01/15/2021  03:04 AM            41,984 iptest3.exe                                                                      
01/15/2021  03:04 AM            41,984 ipython.exe                                                                     
01/15/2021  03:04 AM            41,984 ipython3.exe                                                                     
03/26/2021  02:20 PM            74,752 mako-render.exe                                                                 
01/15/2021  03:04 AM            41,984 mamba.exe                                                                        
09/29/2020  01:01 PM            41,984 menuinst.exe                                                                     
03/26/2021  02:20 PM            74,752 pip.exe                                                                          
03/26/2021  02:20 PM            74,752 pip3.exe                                                                         
11/01/2019  02:30 PM            40,960 pydoc.exe                                                                        
03/26/2021  02:20 PM            74,752 pygmentize.exe                                                                   
04/14/2021  04:15 PM            74,752 tqdm.exe                                                                         
01/15/2021  03:04 AM            41,984 uhd_images_downloader.exe                                                       
01/15/2021  03:04 AM            41,984 uhd_siggen.exe                                                                  
01/15/2021  03:04 AM            41,984 watchmedo.exe                                                                    
03/26/2021  02:20 PM            74,752 wheel.exe                                                                         
           29 File(s)      1,509,376 bytes                                      
              0 Dir(s)  41,682,497,536 bytes free                                                                                                                                                                                              (base) C:\Users\foobar\radioconda\Scripts>gnuradio-companion.exe                                                      
Traceback (most recent call last):                                                                                        
File "C:\Users\foobar\radioconda\Scripts\gnuradio-companion-script.py", line 10, in <module>                       
     sys.exit(main())                                                                                                      
File "C:\Users\foobar\radioconda\lib\site-packages\gnuradio\grc\main.py", line 72, in main                              platform.build_library()                                                                                              
File "C:\Users\foobar\radioconda\lib\site-packages\gnuradio\grc\core\platform.py", line 192, in build_library raise RuntimeError(errstr)                                                                                          
RuntimeError: Failed to find built-in GRC blocks (specifically, the 'options' block). Ensure your GRC block paths are correct and at least one points to your prefix installation:       
           
```

ryanvolz, 2021-04-20:
I can't say I fully understand how the conflict worked here, but "conflict with pothos" is good enough for me. I guess it was taking preference on the executable path.

Only things that are Python scripts with an exe wrapper go in the Scripts directory, so that looks rather normal to me. Other things can be in Library\bin.

closed 2021-04-26 by ryanvolz