radioconda/radioconda-installer #4: Crash on Ubuntu 20.04

jo-rivera, 2021-09-22:
Installed radioconda via command line using installer https://glare.now.sh/ryanvolz/radioconda/radioconda-.*-Linux-x86_64.sh

gnuradio-companion opens the first time, after closing and re-opened again it crashes

[image]

ryanvolz, 2021-09-22:
Thanks for the report! I think this is the same thing we're investigating [here](https://github.com/conda-forge/gnuradio-feedstock/issues/84).

dicta, 2021-10-08:
Looks like this should be fixed here by 32ebbbced4dfccdc40f57d240b0ab1c978ada059 now that the underlying issue is resolved upstream.

ryanvolz, 2021-10-09:
Yes, should be fixed now.

closed 2021-10-09 by ryanvolz