radioconda/radioconda-installer #5: Portaudio support on windows missing?

wlmqljj, 2021-10-09:
It seems portaudio is not working on windows
I have changed my config file 
`[audio]
audio_module = portaudio`
but it just won't work , here is the log 

> Executing: C:\radioconda\python.exe -u E:\radio\FM_STEREO.py
> Traceback (most recent call last):
>   File "E:\radio\FM_STEREO.py", line 367, in <module>
>     main()
>   File "E:\radio\FM_STEREO.py", line 345, in main
>     tb = top_block_cls()
>  File "E:\radio\FM_STEREO.py", line 211, in __init__
>     self.connect((self.audio_source_0, 1), (self.analog_fm_preemph_0_0_0, 0))
>   File "C:\radioconda\lib\site-packages\gnuradio\gr\hier_block2.py", line 37, in wrapped
>     func(self, src, src_port, dst, dst_port)
>   File "C:\radioconda\lib\site-packages\gnuradio\gr\hier_block2.py", line 100, in connect
>     self.primitive_connect(*args)
> ValueError: port number 1 exceeds max of 0
>>> Done (return code 1)

seems like it is using the old windows audio api that does't support stereo audio.( I'm using a 2-port audio source block)
Is there any change to make portaudio work?

Really appreciate！

ryanvolz, 2021-10-09:
Oh interesting, I didn't know that the Windows audio subsystem support was hampered in that way. Portaudio is disabled at build time right now, but it should be easy to enable it. I'll work on that.

wlmqljj, 2021-10-15:
Great! Feel free to commit a release.

closed 2021-10-15 by wlmqljj