radioconda/radioconda-installer #6: Cannot open shared library libasound_module_conf_pulse.so

adrianboyko, 2021-12-12:
Just tried radioconda for the first time and... no luck! I'm using Ubuntu 21.10. `conda list | grep libasound` produces no results.

```
ALSA lib conf.c:3723:(snd_config_hooks_call) Cannot open shared library libasound_module_conf_pulse.so (libasound_module_conf_pulse.so: libasound_module_conf_pulse.so: cannot open shared object file: No such file or directory)
ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM default
gr::log :ERROR: audio_alsa_source0 - [default]: No such file or directory
Traceback (most recent call last):
  File "/home/adrian/Documents/Designs/Skimmer/Spam.py", line 362, in <module>
    main()
  File "/home/adrian/Documents/Designs/Skimmer/Spam.py", line 340, in main
    tb = top_block_cls()
  File "/home/adrian/Documents/Designs/Skimmer/Spam.py", line 261, in __init__
    self.audio_source_0 = audio.source(8000, 'default', True)
RuntimeError: audio_alsa_source

ryanvolz, 2021-12-12:
I think the alsa pulseaudio module comes in the `alsa-plugins` package, so see if you can install that and if it fixes your problem. If it does, it's definitely something I should have radioconda install by default.

adrianboyko, 2021-12-12:
Looks like it's in [libasound2-plugins](https://packages.ubuntu.com/search?mode=exactfilename&suite=impish&section=all&arch=any&keywords=libasound_module_conf_pulse.so&searchon=contents) but `conda install libasound2-plugins` fails with:
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - libasound2-plugins

Current channels:

  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

ryanvolz, 2021-12-12:
That's the package it's in for Ubuntu, but you need the conda package, in which case it *is* [`alsa-plugins`](https://github.com/conda-forge/alsa-plugins-feedstock).
```
mamba install alsa-plugins
```
should get it working for you.

adrianboyko, 2021-12-12:
Ok! Tried `mamba install alsa-plugins` and got:

```
Encountered problems while solving:
  - package gnuradio-3.9.4.0-py39he96b9e0_1 requires alsa-lib >=1.2.3,<1.2.4.0a0, but none of the providers can be installed
```

ryanvolz, 2021-12-12:
Interesting... you should already have `alsa-lib 1.2.3` installed. Maybe mamba has a bug. Try
```
conda install alsa-plugins
```
I'll see if I can try to reproduce the installation failure locally. Regardless, I'll be putting alsa-plugins in the next radioconda release.

ryanvolz, 2021-12-12:
I think `alsa-plugins` is failing to install because of a dependency conflict with `jack` (contrary to what it says about `alsa-lib`), so I'm re-building `alsa-plugins` to see if that will fix it. I'll let you know.

ryanvolz, 2021-12-12:
You should be able to `mamba install alsa-plugins` now.

adrianboyko, 2021-12-12:
Success! Thank you.

ryanvolz, 2022-01-19:
`alsa-plugins` is now included in the installer, and in the latest 2021.01.19 release.

closed 2022-01-19 by ryanvolz