radioconda/radioconda-installer #8: Issue with windows gnuradio osmosdr

sxu912, 2022-02-02:
try to communicate with RTL STR, the following error occurred, seems like it is missing a file. Really appreciate any help you have on this issue. Thanks

During handling of the above exception, another exception occurred:

  File "C:\Users\HSPC1\Documents\GNU radio\RTL_SDR_FM_spectrum_analyzer.py", line 34, in <module>
    import osmosdr
  File "C:\Users\HSPC1\radioconda\lib\site-packages\osmosdr\__init__.py", line 21, in <module>
    from .osmosdr_python import *
ImportError: DLL load failed while importing osmosdr_python: The specified procedure could not be found.

ryanvolz, 2022-02-02:
Thanks for the report. So I can try to reproduce this, what version of radioconda do you have installed, or what is the output of `conda list`?

sxu912, 2022-02-02:
Thanks for getting back to me, below are some of the output from conda list:
  
gnuradio                  3.9.5.0          py38hbbbac86_0    conda-forge
gnuradio-core             3.9.5.0          py38h0268b5b_0    conda-forge

also the radioconda install package is:
radioconda-2022.01.19-Windows-x86_64.exe

ryanvolz, 2022-02-02:
Ok, I'll try it out tomorrow. Is there any chance that you have previously installed rtl-sdr, GNU Radio, or osmosdr through some other method and those are interfering in some way? I only ask because importing osmosdr is one of the things that is tested automatically to see if the package was built correctly, so I'm struggling to come up with a reason for why it would fail for you.

sxu912, 2022-02-02:
yes, I had an older version gnuradio3.7. I am removing this version and reinstall the new one right now. Hopefully it can resolve the issue.

Thanks again.

sxu912, 2022-02-02:
OK, after I removed the older version and did a clean reinstall, the same issue remains. I then looked inside the osmosdr directory inside siteand could not find the file:

\radioconda\Lib\site-packages\osmosdr\.osmosdr_python

[image]

ryanvolz, 2022-02-02:
The file it needs to load is the one that is there, `osmosdr_python.cp38-win_amd64.pyd`. I just tried this on my installation and I had no issues, so we have to figure out what is different for you. Thoughts:

- What version of Windows are you using? I have Windows 10, but it should work for 8+.
- Are you using a 32-bit version of Windows? This is only built for 64-bit.
- Are you running from an activated environment? I.e. either from the "radioconda prompt" or a command line after having run `conda activate base`?

sxu912, 2022-02-02:
Thanks Ryan.

I am using Win10/64-bit. Yes, I activated radioconda according the instructions from GNU windows installation guide,

I also tried to run the python code from the command line, and got the following message, which is different from running from GUI. I don't know if this error message can offer a hint.

Thanks

DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives from distutils.version import StrictVersion
Traceback (most recent call last):
  File "C:\Users\HSPC1\Documents\GNU radio\RTL_SDR_FM_spectrum_analyzer.py", line 23, in <module>
    from PyQt5 import Qt
ModuleNotFoundError: No module named 'PyQt5'

ryanvolz, 2022-02-02:
It's something else to investigate at least. I can see that I have `pyqt 5.12.3` installed in the conda environment after doing a `conda list`, and I assume it's the same for you?

Do you get a more informative message if you run
```
python -c "import PyQt5"
python -c "from PyQt5 import Qt"
```
from the command line?

What does your `PATH` look like? Try `echo %PATH%`.

sxu912, 2022-02-03:
Hi Ryan, 

I do see I have pyqt5.12.3. I tried the following commands from the command line and get nothing back, so I assume this is normal. 

python -c "import PyQt5" 
python -c "from PyQt5 import Qt"

But I ran echo %PATH", I see Python27 path, but Python38 path is not there. Maybe this is the problem. I should modify the PATH for Python38.

Thanks

ryanvolz, 2022-02-08:
Did fixing the path resolve your issue?

I just made a release that should fix this problem (or something like it) when the cause is interfering environment variables set by PothosSDR. If it's not just the path and you also have PothosSDR installed, then give 2022.02.07 a try.

sxu912, 2022-02-08:
Thanks Ryan. I changed the environment variable for python, but didn't fix osmosdr problem. I will try your new version on a different computer, I think maybe it is my computer has too much stuff installed. I also ordered a limesdr, really hope gnuradio will work with limesdr. I will keep you updated.

ryanvolz, 2024-05-09:
I think this is either resolved, or issues with Qt are now tracked at https://github.com/ryanvolz/radioconda/issues/78.

closed 2024-05-09 by ryanvolz