radioconda/radioconda-installer #10: Return code 3221225501

riocorre, 2022-02-12:
Hi,

I have installed GNU Radio 3.9.5.0 on Windows 10. I have a simple configuration of a RTL-SDR sink and QT GUI Sink. 

The error makes end the program with this code:  Return code 3221225501
It also happens with QT GUI Frequency Sink.

It does not happen with QT Time Sink.

Do you have some hints about it?

Thanks!

Enrique

ryanvolz, 2022-02-12:
Hi Enrique!

I take this to mean that your flowgraph does not run at all, immediately crashing with that error code? Does it say anything else before that happens?

Some searching seems to indicate that that error code is related to being unable to load a DLL. How are you running GNU Radio Companion? (e.g. from the Start Menu entry called "GNU Radio Companion (radioconda)", or by running `gnuradio-companion` from the "Conda Prompt (radioconda)"). It's possible that it can't find a DLL because the conda environment has not been activated properly by running using one of those methods.

riocorre, 2022-02-13:
Hi Ryan,

thank you for your email.

I have the same error If I run the application from the Start Menu or  by
the Condia Prompt.

I add some images.

[image: image.png]
[image: image.png]
[image: image.png]

[image: image.png]

Thanks!

Enrique

ryanvolz, 2022-02-13:
Hi Enrique,

The images didn't come through Github's system. I think you'll have to use the website to post them to the issue.

riocorre, 2022-02-13:
Thank you Ryan,

here you have the images,

Cheers,
[image]
[image]
[image]
[image]

Enrique

riocorre, 2022-02-14:
Hi, Ryan,
 here are the images.
[image]
[image]
[image]
[image]

riocorre, 2022-02-14:
Hi Ryan,

I have uploaded the images at Github, did you received them?

Enrique

ryanvolz, 2022-02-14:
Hi Enrique,

This is a tough one. That error code is not much to go on, and that's the only clue we have besides that fact that it works for the Qt Time sink but not the GUI sink. I tried your exact flowgraph and it worked fine for me, so that doesn't help.

Maybe we can figure out what is different about your system, specifically related to Qt since that seems to be a culprit. Do you have any other versions of GNU Radio installed (e.g. GNCDevelopment, PothosSDR), or have they been installed in the past? Interference with old/other installations has sometimes been a problem. How about Qt being installed somewhere else as well? Any other applications installed that you know use Qt and may have brought and incompatible version? Anything else special about your setup that might be relevant?

Failing progress from that, maybe the next step would be to launch `ipython` from the Conda Prompt and then manually enter the Python script produced by your failing flowgraph line-by-line, until we can see what line causes the segmentation fault.

riocorre, 2022-02-14:
Thank you Ryan.

It is the first time I installed GNU radio on this computer.

Enrique

riocorre, 2022-02-15:
> I think running the flowgraph-generated script manually line-by-line is
> your next best bet for narrowing this down. That should at least tell us
> what line is causing the error, and maybe it gives you a better error
> message. I suggest using an ipython interactive session.

Thank you Ryan.

I will let it as is by the moment.

Enrique

ryanvolz, 2023-01-19:
I think this is better summarized in #25.

closed 2023-01-19 by ryanvolz