radioconda/radioconda-installer #11: pmt not accessible from File Meta Sink block

tonycolo, 2022-02-13:
I'm unable to use the File Meta Sink in GRC 3.9.5 Release 2022.02.07 due to GRC not being able to access pmt.

The default value for Extra Dict causes an error:

`Param - Extra Dict.(extra_dict):
	Value "pmt.make_dict()" cannot be evaluated:
	name 'pmt' is not defined`

Using Conda Prompt, I'm able to import pmt and invoke pmt.make() without a problem.
For some reason GRC can't access the Polymorphic Types module, at least through the File Meta Sink block.  I tried a Variable block and it wouldn't work either.

ryanvolz, 2022-02-13:
Hi! Since it can load `pmt` when using a Conda Prompt, do you have any success if you run `gnuradio-companion` from the Conda Prompt? Otherwise, are you launching GRC from the "GNU Radio Companion (radioconda)" entry in the Start Menu?

tonycolo, 2022-02-13:
I was running from Conda Prompt and the same problem exists when running from any shortcut / start menu entry.   I start Conda Prompt and then invoke   Scripts\gnuradio-companion.exe.

ryanvolz, 2022-02-14:
I suppose not many (any) people have tried the File Meta Sink yet, and it was just created in a rush before the release. Definitely file a bug upstream so they know!

tonycolo, 2022-02-14:
I'm not sure what you mean.  The first commit of the File Meta Sink block implementation [dates to 2012](https://github.com/gnuradio/gnuradio/commit/461ece56b36a44b2405282630157739c7f9a26ba), and the issue above notes the pmt import about a year ago.
Thank you for putting together the distribution.   I have been trying basic things with writing basic FFT output on the road to using my LimeSDR  and RTL-SDR for amateur radioastronomy.  I see your digital_rf and perhaps that will be much better than anything I could do.

ryanvolz, 2022-02-14:
Oh sorry, I completely overlooked the issue you linked and also simultaneously confused this with the new SigMF block. So I guess this was "fixed" with documentation, but it seems like a code solution would be better. I'll see about putting in a pull request. Thanks for doing all the debugging, it made my job easy!

The Digital RF package could be useful if you plan on collecting data the way that we do at my work, which is basically running the radio all the time and writing the data continuously to disk. Then we do some automated processing, and the data is there if we want to do more digging. Eventually we delete the oldest data in a pseudo-ringbuffer fashion (Digital RF has tools for that). If that's not the kind of data collection you want to do and instead want to target fixed-length experiments, then you might be happier working with the File Meta Sink or the new SigMF stuff.

Good luck with your radio astronomy, and I'm glad that radioconda could be useful for it! Enabling this kind of thing was my motivation, anyway.