radioconda/radioconda-installer #12: QT GUI Widgets not working in 2022.01.20.dev

tonycolo, 2022-02-15:
File "tutorial1.py", line 124, in __init__
    self._qtgui_freq_sink_x_1_win = sip.wrapinstance(self.qtgui_freq_sink_x_1.pyqwidget(), Qt.QWidget)
AttributeError: 'gnuradio.qtgui.qtgui_python.freq_sink_f' object has no attribute 'pyqwidget'

Same error for QT GUI Time sink.
Windows 10, fresh install of 2022.01.20.dev

ryanvolz, 2022-02-15:
I think this is a GNU Radio 3.9 -> 3.10 change, and your `tutorial1.py` flowgraph needs to be recompiled with 3.10. If you don't have a GRC file to regenerate the .py file from, then I think all you should have to do is change `pyqwidget()` in that line to `qwidget()`.

tonycolo, 2022-02-16:
This is a flow graph created in GRC.  I start with a blank flowgraph in 3.10.0 (rather than loading a previous one) and see the same result.

ryanvolz, 2022-02-16:
That sounds like it's loading the block's YAML file from a different, older GNU Radio version. What is the output of
```
gnuradio-config-info --print-all
```
?

tonycolo, 2022-02-16:
(base) C:\Users\tonyb>gnuradio-config-info --print-all ?
C:\Program Files\PothosSDR
C:\Program Files\PothosSDR\etc
C:\Program Files\PothosSDR\etc\gnuradio\conf.d
C:\Users\tonyb\AppData\Roaming\.gnuradio
Wed, 19 Jan 2022 20:08:05Z
python-support;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* portaudio;* windows;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;gr-wavelet;gr-zeromq;gr-network;gr-soapy
3.10.0.0
1916
1916
C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe:::/MD /O2 /Ob2 /DNDEBUG /DWIN32 /D_WINDOWS /W3
C:/Program Files (x86)/Microsoft Visual Studio/2019/Enterprise/VC/Tools/MSVC/14.16.27023/bin/HostX64/x64/cl.exe:::/MD /O2 /Ob2 /DNDEBUG /DWIN32 /D_WINDOWS /W3 /GR /EHsc
2.9.0

ryanvolz, 2022-02-16:
Yeah, PothosSDR stomps on other GNU Radio installations. Run this from the Conda Prompt
```
conda env config vars set GR_PREFIX= GRC_BLOCKS_PATH= UHD_PKG_PATH= VOLK_PREFIX= 
```
and then restart the prompt to run GRC. That should fix it, but maybe only for running from the prompt and not the Start Menu icon. Uninstalling PothosSDR would guarantee that it doesn't interfere, but I won't assume that you don't need it.

tonycolo, 2022-02-16:
Thank you for helping me resolve.  I had to rename/hide the PothosSDR directory.

closed 2022-02-16 by tonycolo