radioconda/radioconda-installer #13: errors using "conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda"

davidsaroff, 2022-02-28:
With an updated conda on a ubuntu 18.04 I get errors using 
"conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda"

1) in a terminal window
$ conda update conda

2)$ conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda

this error report results
# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/exceptions.py", line 1080, in __call__
        return func(*args, **kwargs)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/cli/main.py", line 84, in _main
        exit_code = do_call(args, p)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/cli/conda_argparse.py", line 83, in do_call
        return getattr(module, func_name)(args, parser)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/cli/main_create.py", line 41, in execute
        install(args, parser, 'create')
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/cli/install.py", line 315, in install
        handle_txn(unlink_link_transaction, prefix, args, newenv)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/cli/install.py", line 344, in handle_txn
        unlink_link_transaction.execute()
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/core/link.py", line 245, in execute
        self.verify()
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/common/io.py", line 88, in decorated
        return f(*args, **kwds)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/core/link.py", line 222, in verify
        self.prepare()
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/core/link.py", line 214, in prepare
        stp.neutered_specs)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/core/link.py", line 289, in _prepare
        for prec, pcrec in zip(link_precs, pkg_cache_recs_to_link))
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/core/link.py", line 289, in <genexpr>
        for prec, pcrec in zip(link_precs, pkg_cache_recs_to_link))
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/gateways/disk/read.py", line 90, in read_package_info
        paths_data = read_paths_json(epd)
      File "/home/dsaroff/anaconda3/lib/python3.7/site-packages/conda/gateways/disk/read.py", line 160, in read_paths_json
        data = json.load(paths_json)
      File "/home/dsaroff/anaconda3/lib/python3.7/json/__init__.py", line 296, in load
        parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
      File "/home/dsaroff/anaconda3/lib/python3.7/json/__init__.py", line 348, in loads
        return _default_decoder.decode(s)
      File "/home/dsaroff/anaconda3/lib/python3.7/json/decoder.py", line 337, in decode
        obj, end = self.raw_decode(s, idx=_w(s, 0).end())
      File "/home/dsaroff/anaconda3/lib/python3.7/json/decoder.py", line 355, in raw_decode
        raise JSONDecodeError("Expecting value", s, err.value) from None
    json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

`$ /home/dsaroff/anaconda3/bin/conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda`

  environment variables:
                 CIO_TEST=<not set>
          COMPIZ_BIN_PATH=/usr/bin/
        CONDA_DEFAULT_ENV=base
                CONDA_EXE=/home/dsaroff/anaconda3/bin/conda
             CONDA_PREFIX=/home/dsaroff/anaconda3
    CONDA_PROMPT_MODIFIER=(base)
         CONDA_PYTHON_EXE=/home/dsaroff/anaconda3/bin/python
               CONDA_ROOT=/home/dsaroff/anaconda3
              CONDA_SHLVL=1
           CURL_CA_BUNDLE=<not set>
            DEFAULTS_PATH=/usr/share/gconf/unity.default.path
GNOME_SESSION_XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
          LD_LIBRARY_PATH=/home/dsaroff/presto/lib:/usr/local/cuda-10.1/lib64
           MANDATORY_PATH=/usr/share/gconf/unity.mandatory.path
                     PATH=/home/dsaroff/anaconda3/bin:/usr/local/cuda-10.1/bin:/home/dsaroff/ana
                          conda3/bin:/home/dsaroff/anaconda3/condabin:/home/dsaroff/anaconda2/bi
                          n:/home/dsaroff/bin:/home/dsaroff/.local/bin:/usr/local/sbin:/usr/loca
                          l/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/
                          bin
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>
            XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
         XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0

     active environment : base
    active env location : /home/dsaroff/anaconda3
            shell level : 1
       user config file : /home/dsaroff/.condarc
 populated config files : /home/dsaroff/.condarc
          conda version : 4.11.0
    conda-build version : 3.20.5
         python version : 3.7.8.final.0
       virtual packages : __cuda=11.0=0
                          __linux=4.15.0=0
                          __glibc=2.27=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/dsaroff/anaconda3  (writable)
      conda av data dir : /home/dsaroff/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/ryanvolz/linux-64
                          https://conda.anaconda.org/ryanvolz/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/dsaroff/anaconda3/pkgs
                          /home/dsaroff/.conda/pkgs
       envs directories : /home/dsaroff/anaconda3/envs
                          /home/dsaroff/.conda/envs
               platform : linux-64
             user-agent : conda/4.11.0 requests/2.27.1 CPython/3.7.8 Linux/4.15.0-118-generic ubuntu/18.04.5 glibc/2.27
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.

ryanvolz, 2022-03-01:
I don't see an error like this, but I did notice that the most recent few releases are no longer installable like this through the `radioconda` metapackage. This is because one of the dependencies has been marked as broken by conda-forge maintainers and thus can't be installed by `conda` or `mamba`. I'm in the process of making a new release, so once that is out, try this again and see if you have better results. It might also be instructive to try the same command with `mamba` instead of `conda`.

davidsaroff, 2022-03-12:
Ryan,

I'm trying the windows installer that I find here
https://github.com/ryanvolz/radioconda/releases/download/2022.02.28/radioconda-2022.02.28-Windows-x86_64.exe

It works. gqrx launched from the windows start menu, and from the conda
prompt finds the pluto, and local fm channels can be tuned.

Thanks!

David Saroff
KM4DCQ

closed 2022-03-14 by ryanvolz