radioconda/radioconda-installer #14: gr_lora-sdr Building OOT modules to use with conda-installed GNU Radio

ewwll, 2022-03-08:
The following error occurs when constructing GR-LORA_SDR using Windows VS2019 including MSVC V141
[1/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\crc_verif_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/crc_verif_impl.cc.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib\..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\crc_verif_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\crc_verif_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib\..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
[2/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\gray_enc_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/gray_enc_impl.cc.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib\..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\gray_enc_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc(15): error C2440: “return”: 无法从“std::shared_ptr<gr::lora_sdr::gray_enc_impl>”转换为“boost::shared_ptr<gr::lora_sdr::gray_enc>”
D:\360\lora\gr-lora_sdr-master\lib\gray_enc_impl.cc(15): note: 无构造函数可以接受源类型，或构造函数重载决策不明确
[3/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\header_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/header_impl.cc.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib\..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\header_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(16): error C3646: “m_cnt_nibbles”: 未知重写说明符
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(16): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(17): error C3646: “m_cnt_header_nibbles”: 未知重写说明符
d:\360\lora\gr-lora_sdr-master\lib\header_impl.h(17): error C4430: 缺少类型说明符 - 假定为 int。注意: C++ 不支持默认 int
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(16): error C2440: “return”: 无法从“std::shared_ptr<gr::lora_sdr::header_impl>”转换为“boost::shared_ptr<gr::lora_sdr::header>”
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(16): note: 无构造函数可以接受源类型，或构造函数重载决策不明确
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(35): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(82): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(89): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(92): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(100): warning C4805: “|”: 在操作中将类型“int”与类型“bool”混合不 安全
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(110): warning C4805: “|”: 在操作中将类型“int”与类型“bool”混合不 安全
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(119): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(121): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(122): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(132): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(140): error C2065: “m_cnt_nibbles”: 未声明的标识符
D:\360\lora\gr-lora_sdr-master\lib\header_impl.cc(141): error C2065: “m_cnt_header_nibbles”: 未声明的标识符
[4/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\frame_sync_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/frame_sync_impl.cc.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib\..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\frame_sync_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\frame_sync_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib\..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
[5/24] Building CXX object lib\CMakeFiles\gnuradio-lora_sdr.dir\header_decoder_impl.cc.obj
FAILED: lib/CMakeFiles/gnuradio-lora_sdr.dir/header_decoder_impl.cc.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DBOOST_ALL_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFMT_SHARED -DGR_MPLIB_MPIR -DGR_PERFORMANCE_COUNTERS -DMS_WIN64 -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_SHARED_LIB -DWIN32_LEAN_AND_MEAN -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_SCL_SECURE_NO_WARNINGS -D_USE_MATH_DEFINES -D_WIN32_WINNT=0x0502 -Dgnuradio_lora_sdr_EXPORTS -ID:\360\lora\gr-lora_sdr-master\lib\..\include -ID:\360Downloads\anaconda\envs\gnuradio\Library\include -ID:\360Downloads\anaconda\envs\gnuradio\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\header_decoder_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\header_decoder_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
[... cut ...] include /DWIN32 /D_WINDOWS /W3 /GR /EHsc -std=c++11 /MD /O2 /Ob2 /DNDEBUG /wd4251 /wd4275 -std:c++17 /showIncludes /Folib\CMakeFiles\gnuradio-lora_sdr.dir\add_crc_impl.cc.obj /Fdlib\CMakeFiles\gnuradio-lora_sdr.dir\ /FS -c D:\360\lora\gr-lora_sdr-master\lib\add_crc_impl.cc
cl: 命令行 warning D9002 :忽略未知选项“-std=c++11”
D:\360\lora\gr-lora_sdr-master\lib\..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
ninja: build stopped: subcommand failed.

ryanvolz, 2022-03-08:
Thanks for the report! The key part is the following:
```
D:\360\lora\gr-lora_sdr-master\lib..\include\lora_sdr/utilities.h(9): fatal error C1083: 无法打开包括文件: “sys/resource.h”: No such file or directory
```
That header file does not exist on Windows (Unix only), so the source code would have to be changed to work around that. From what I can tell, there is no trivial drop-in replacement. Whatever that header is needed for, it will need to be ported to use a Windows equivalent.

ewwll, 2022-03-09:
> 

Thank you for your answer

closed 2023-01-19 by ryanvolz