radioconda/radioconda-installer #15: bladeRF support

nhw76, 2022-03-09:
It would be really great if radioconda could package support for the nuand BladeRF, either via SoapySDR or a standalone module.

ryanvolz, 2022-03-09:
Agreed, that's probably the biggest hardware hole right now!

nhw76, 2022-03-09:
I'm pretty clueless about conda; it's not an ecosystem I've had on my radar before, but it looks like the gnuradio-osmosdr package over here ... https://github.com/conda-forge/gnuradio-osmosdr-feedstock/ ... builds with support for bladeRF turned off; not sure if there's another upstream dependency that's missing or whether it's as simple as changing an OFF to an ON.

In any case, I'm happy to help (even if only by testing). Let me know how I can help.

ryanvolz, 2022-03-09:
Right, there would be three steps:

1. Add a `bladerf` package for the hardware driver.
2. Add a `soapysdr-module-bladerf` package that builds the Soapy module and depends on `bladerf`.
3. Build `gnuradio-osmosdr` with bladeRF support and depend on `bladerf`.

Modeling on the existing hardware packages, this shouldn't be too hard. It will probably take me a couple hours to put together the recipes and make sure they can build (this depends heavily on how well the upstream is currently supporting macOS and Windows builds, which can sometimes throw in a few wrenches). Then it would probably be about a week to get it merged into conda-forge, and then finally I could include it in radioconda. I might have enough time to give it a try soon, but no guarantees! Then I'll come back and ask for some testing.

ryanvolz, 2022-03-18:
This was a little more challenging than I anticipated, but I have recipes now pending review/merge at https://github.com/conda-forge/staged-recipes/pull/18397. We're now at the mercy of conda-forge's volunteer staged-recipes reviewers to get the feedstocks created and have the built packages uploaded. Could be about ~1 week, but sometimes it can be much faster or slower. After that, I'll ping again to request some testing!

florianjomrich, 2022-04-02:
I am also really missing this out - to get my bladeRF working under Windows was unsuccessful. Under Ubuntu as well and the Live Images of GNU Radio with bladeRF are not maintained anymore. 
So I also would highly appreciate radioconda to support bladeRF as well :).

ryanvolz, 2022-04-15:
Released in 2022.04.15

closed 2022-04-15 by ryanvolz