radioconda/radioconda-installer #16: gr-lora_sdr using Cmake in the WSL-Ubuntu

ewwll, 2022-03-14:
I  compiled this project absout gr-lora_sdr using Cmake in the WSL-Ubuntu environment and now the following problem occurs, all the dependent libraries have been installed
-- The CXX compiler identification is GNU 9.4.0
-- The C compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Conda installed GNU Radio. Setting CMAKE_INSTALL_PREFIX to /root/anaconda3/envs/gnuradio
-- Build type not specified: defaulting to release.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'gmp'
-- Found gmp, version 6.2.1
-- Found GMP: /usr/local/lib/libgmpxx.so
-- Using GMP.
-- Found MPLIB: /usr/local/lib/libgmpxx.so
-- Found Boost: /root/anaconda3/envs/gnuradio/lib/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread
-- Found Volk: Volk::volk
-- User set python executable /root/anaconda3/envs/gnuradio/bin/python
-- Found PythonInterp: /root/anaconda3/envs/gnuradio/bin/python (found version "3.8.12")
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found suitable exact version "3.8.12")
-- Checking for module 'sndfile'
-- Found sndfile, version 1.0.28
-- Found SNDFILE: /usr/local/lib/libsndfile.so
-- Checking for module 'fftw3f >= 3.0'
-- Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Found Git: /usr/bin/git
-- Extracting version information from git describe...
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- Using install prefix: /root/anaconda3/envs/gnuradio
-- Building for version: https://github.com/tapparelj/gr-lora_sdr/commit/ca195cb14a0232c5502963a7593f372c3713aa88 / 1.0.0git
-- No C++ unit tests... skipping
-- Found SWIG: /usr/bin/swig4.0 (found version "4.0.1")
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found version "3.8.12")
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of size_t
-- Check size of size_t - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Found PythonLibs: /root/anaconda3/envs/gnuradio/lib/libpython3.8.so (found suitable version "3.8.12", minimum required is "2")
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Success
Traceback (most recent call last):
File "", line 2, in
TypeError: Unicode-objects must be encoded before hashing
CMake Error at cmake/Modules/UseSWIG.cmake:199 (add_executable):
The target name "/home/ewwll/gr-lora_sdr/build/swig/.cpp" is reserved or
not valid for certain CMake features, such as generator expressions, and
may result in undefined behavior.
Call Stack (most recent call first):
cmake/Modules/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
cmake/Modules/GrSwig.cmake:173 (SWIG_ADD_MODULE)
swig/CMakeLists.txt:49 (GR_SWIG_MAKE)

CMake Error at cmake/Modules/GrSwig.cmake:211 (GR_LIBTOOL):
Unknown CMake command "GR_LIBTOOL".
Call Stack (most recent call first):
swig/CMakeLists.txt:54 (GR_SWIG_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/home/ewwll/gr-lora_sdr/build/CMakeFiles/CMakeOutput.log".
See also "/home/ewwll/gr-lora_sdr/build/CMakeFiles/CMakeError.log".

ryanvolz, 2022-03-14:
At a high level, it looks like `gr-lora_sdr`'s CMake code was not really updated to align with GNU Radio 3.8; the version of `cmake/Modules/UseSWIG.cmake` that it has is from the GNU Radio 3.7 era. Updating that (probably a big job, and I'm not saying it's your job) would almost certainly improve things.

At a lower level, this is the key error:
```
TypeError: Unicode-objects must be encoded before hashing
```
which happens when executing the Python code from [these lines](https://github.com/tapparelj/gr-lora_sdr/blob/master/cmake/Modules/UseSWIG.cmake#L176-L181).

I think that is ultimately a problem coming from using Python 3 with code that was originally written to work with Python 2. That code snippet probably needs to be upgraded to be compatible with Python 3 by doing what the error says and `.encode()`ing the string before `hashlib.md5()` is evaluated. But if that were the full problem, then I wouldn't think this would work for anybody with Python 3, yet gr-lora_sdr claims Python 3 compatibility so I don't know what is unique about your setup. Perhaps is has to do with having Unicode characters in your file paths.

ewwll, 2022-03-25:
> At a high level, it looks like `gr-lora_sdr`'s CMake code was not really updated to align with GNU Radio 3.8; the version of `cmake/Modules/UseSWIG.cmake` that it has is from the GNU Radio 3.7 era. [...]

closed 2023-01-19 by ryanvolz