radioconda/radioconda-installer #17: python 3.10?

cessna85, 2022-03-26:
Are there any plans to update python to 3.10.x?

ryanvolz, 2022-03-26:
Yeah, I can change it for the next release. The only thing keeping it at Python 3.8 is compatibility with Windows 7 (other platforms are already at Python 3.10), and there probably aren't enough Windows 7 users to justify that.

Just curious: what features from Python 3.10 are you wanting to have?

cessna85, 2022-03-26:
Ahhh... That makes sense.
I am looking to use the new [pattern matching syntax](https://peps.python.org/pep-0636/) 
I was able to upgrade to python to 3.10 using radioconda's environment without any issues so it just made me curious to if there was a particular package that had issues with 3.10.

ryanvolz, 2022-04-15:
Released in 2022.04.15

closed 2022-04-15 by ryanvolz