radioconda/radioconda-installer #18: After install the radioconda, I can't open the GNUradio

blaticslm, 2022-04-07:
When finish the installation, I click the GNUradio icon, and the cmd window shows up for less than a second and then disappears. How to solve this issue? 

[image]

The picture above is showing the crash. 

Really thanks for your response!

ryanvolz, 2022-04-08:
I don't see anything in your screenshot, but I can conclude from the description that GRC is crashing when you launch it. In order to get more information about what's causing the crash, try the following:

1. Launch the "Conda Prompt" from the Start menu.
2. Run the command `gnuradio-config-info --print-all` and post the output here.
3. Run the command `gnuradio-companion` (expecting it will crash) and post the output here.

blaticslm, 2022-04-08:
> I don't see anything in your screenshot, but I can conclude from the description that GRC is crashing when you launch it. In order to get more information about what's causing the crash, try the [...]

I think I made it! I forget to set up the conda environment for gnuradio. Now it works.

ryanvolz, 2022-04-08:
Great!

closed 2022-04-08 by ryanvolz

reverseideol0gy, 2022-05-08:
> I think I made it! I forget to set up the conda environment for gnuradio. Now it works.

i can't run conda Prompt too 
[image]

้how to set conda environment for gnuradio can you share how to fix for me i have this problem too.

blaticslm, 2022-05-08:
https://wiki.gnuradio.org/index.php/CondaInstall#Installation_using_conda

I think you just need to uninstall the one you have and delete all its relevant files. Then you can follow the steps from the above website. I redo it and eventually works for me.

ryanvolz, 2022-05-08:
@reverseideol0gy You have Python 2.7 installed in `C:\\Python27` and that is clobbering everything the conda installation is trying to do. I don't know why that's happening, but surely you could fix it by uninstalling that other version of Python. If that's not possible, then maybe reinstalling radioconda and choosing to set it as your system Python would override the other installation, but that might be futile.