radioconda/radioconda-installer #21: IDP.Generic Threat

PE2016, 2022-04-25:
When trying to install radioconda 3.10.2.0 on Windows 10, the AVG antivirus alerted me for an IDP.Generic threat. Is this a real threat?
Even when I accpted, AVG didn't allow the installation of gdbus.exe
Please advise, thanks

ryanvolz, 2022-04-26:
You'll have to trust me and the conda-forge community, but we're not putting a virus in there (at least intentionally - it would be a big security breach if a nefarious actor snuck something in!).

`gdbus.exe` is part of the `glib-tools` package which is built from the [glib feedstock](https://github.com/conda-forge/glib-feedstock). I don't know why AVG antivirus thinks it's a threat, but it is a normal part of glib which is depended on by the Gtk GUI toolkit among other things. If you can override AVG, then I would suggest that.

PE2016, 2022-04-27:
Hello Ryan
I added an exception to suppress the virus alerts  and it installed. Now,
where do I find support and updated steps to communicate with the USRP 210N?
Attached is what I got when running conda info; am I missing something?
Thank you,
Ramez

ryanvolz, 2022-04-27:
Github scrubbed your attachment, so I can't see what your `conda info` shows, but it probably doesn't matter. Now that you have a complete installation, you will have the `uhd` package installed. There are no special steps needed for using a USRP N210 with radioconda on Windows, so you should follow the setup specified by NI/Ettus and do any troubleshooting that they recommend. Running `uhd_find_devices` is a great way to test whether your device is accessible. If you don't see it, you probably need to set up your IP address or the device's IP address and check your connections.

PE2016, 2022-04-28:
Ryan
Thank you for your prompt response. Find devices worked, where can I find a list of commands?
Best
Ramez

ryanvolz, 2022-04-28:
What kind of commands? Arguments for `uhd_find_devices`? Other UHD-related commands?

ryanvolz, 2022-04-28:
All of the following programs can be executed as commands:
```
uhd_cal_rx_iq_balance
uhd_cal_tx_dc_offset
uhd_cal_tx_iq_balance
uhd_config_info
uhd_find_devices
uhd_image_loader
uhd_images_downloader
uhd_usrp_probe
usrp2_card_burner
```

Besides `uhd_find_devices`, I find `uhd_image_loader`, `uhd_images_downloader`, and `uhd_usrp_probe` to be the most commonly useful.

closed 2023-01-19 by ryanvolz