radioconda/radioconda-installer #24: /? Text Mistake

Calyso, 2022-07-31:
The text for /NoRegistry (brought up using *.exe /?) lists AllUsers and JustMe both as 0. Found in the 2022.06.24 release and would be considered a cosmetic issue. 

[image]

ryanvolz, 2022-08-01:
Thanks for the report. All of that text is inherited from [conda constructor](https://github.com/conda/constructor), so it might take some digging to get it changed. I can look into it though.

ryanvolz, 2022-08-01:
Oh, I get what this is trying to say now: for either installation type (AllUsers and JustMe), the default for /NoRegistry is 0 (not enabled). So I suppose the text is correct, but it would be clearer just to list "default: 0".

closed 2023-01-19 by ryanvolz