radioconda/radioconda-installer #27: GNU radio companion does not start

ottomc2, 2022-09-22:
I installed it recently and I have an icon but the app does not want to start.
a prompt window comes up, few lines appear in it, and everything disappears in few seconds.
Latest revision radioconda 2002.6.24.0
downloaded from here: 
https://github.com/ryanvolz/radioconda
There is the only version for windows available there:
radioconda-Windows-x86_64.exe
installed successfully
Typing like "gnuradio-companion &" in a prompt window does not help much.
En error message comes: cannot import gnu radio

ottomc2, 2022-09-28:
it does not seem to like % after PYTHONPATH
also, echo -command is not recognized
the location of .exe file looks correct otherwise?

ryanvolz, 2022-09-28:
> (base) C:\Users\ernst.otto>python.exe Python 3.10.5 | packaged by conda-forge | (main, Jun 14 2022, 06:57:19) [MSC v.1929 64 bit (AMD64)] on win32 Type "help", "copyright", "credits" or "license" for more information.
> 
> SyntaxError: invalid syntax
> 
> > > > echo %PYTHONPATH%
> > > > File "", line 1
> > > > echo %PYTHONPATH%
> > > > ^
> > > > SyntaxError: invalid syntax
> > > > echo %PYTHONPATH
> > > > Traceback (most recent call last):
> > > > File "", line 1, in 
> > > > NameError: name 'echo' is not defined

Oh, the `echo %PYTHONPATH%` line is meant to be executed from the Conda prompt and not within the Python interpreter.

> the location of .exe file looks correct otherwise?

Yeah, that looks good. And running `python` launches the correct conda-forge version. I'm stumped.

After launching the Python interpreter (`python.exe`), can you report what the following code returns?
```
import sys
print(sys.path)
```

ryanvolz, 2022-09-29:
Thanks. My installation reports essentially the same, but without the last 3 entries (`\win32`, `\win32\lib`, `\Pythonwin`). Since they come at the end and the other ones look right, I don't think they would have any effect. But it's still curious to me.

You said `import gnuradio` errored. How about some other things, like `import math` and `import numpy`?

ottomc2, 2022-09-29:
>>> import math
>>>

ottomc2, 2022-09-29:
> import math
>

ottomc2, 2022-09-29:
this interface does not like >

ryanvolz, 2022-09-29:
So the standard library package built into Python (`math`) works, but not an external package (`numpy`). I don't know what that tells us and will have to think about it for a little bit.