radioconda/radioconda-installer #28: gnuradio-build-deps: Visual Studio not found

Modiug, 2022-09-22:
I would like to build OOT modules under WIndows 10.
After "conda install gnuradio-build-deps", the compiler is not found. I see that the script is looking for a Visual Studio 2019. I couldn't figure out which condaforge paket to look at for finding out more about this detection process. Maybe more Visual Studio locations can easily be added.
I tried with the latest Visual Studio Community, which is installed in a 2022 directory. The Visual Studio Professional on my work PC is also not found. I would like to dig deeper, but couldn't figure out where to look. Can somebody give me a hint? Maybe I can then provide you with a more detailed issue at the right place. :-)

ryanvolz, 2022-09-22:
You should be fine to use any variant of Visual Studio. The key is to install the "MSVC v141 - VS2017 C++ x64/x86 build tools (v14.16)" component under the list of "Individual Components" when modifying your Visual Studio installation. There's also the VS2019 equivalent that you might need instead, depending on how new your `gnuradio-build-deps` package is. I'll have to dig out the exact name of what that component is called.

ryanvolz, 2022-09-22:
I can't confirm with a local installation yet, but I think the VS2019 component to install withing VS2022 is the similarly-named "MSVC v142 - VS 2019 C++ x64/x86 build tools".

ryanvolz, 2022-09-22:
I'll also add that I don't think that conda-forge, which supplies the compiler setup scripts, fully supports Visual Studio 2022 yet. More specifically, it might have trouble finding the VS2019 compiler within a VS2022 installation even when you have it. That's probably where you'll end up even with everything installed correctly on your end, and then the fact that it doesn't work is not your fault! Hopefully conda-forge fixes it soon, and I think they will. But until then, you might have to install Visual Studio 2019 instead of 2022 with the 2019 compiler package to get it working.

Modiug, 2022-09-24:
Uninstalling VS2022 Community and installing VS2019 Community instead solved the issue so far. VisualStudio is now properly found. I still have to check what's going on with my work laptop where I have some VS Professional installed. At least I am a step further. Thanks!

ryanvolz, 2022-09-26:
Glad it worked!

It looks like VS2022 support is waiting on https://github.com/conda-forge/vc-feedstock/pull/46. I'll keep this open until that is merged so I remember to track it.

Modiug, 2022-09-27:
Sounds good!

In "...\AppData\Local\radioconda\etc\conda\activate.d\" I only see a "vs2019_compiler_vars.bat". So I guess, it really only works with VS2019. I guess there is also some dependency on how Python for Radioconda was built?

ryanvolz, 2022-09-27:
Right, `gnuradio-build-deps` installs the same compiler package that `gnuradio` was built with, which is now VS2019. (It was VS2017 until recently, I think even the previous release.) Thus the need for having the VS2019 installed externally. What's missing is that it should be possible to have VS2022 with the VS2019 build tools installed, but that activation script needs to be updated to support that.

ryanvolz, 2022-10-27:
VS2022 should work now once radioconda is updated to include the latest `vs2015_runtime` package. I'll close this with the next release.