radioconda/radioconda-installer #29: SDRPlay support

jmillo, 2022-10-12:
I am trying to use radioconda wit a RSP1A device from SDRPlay.
I use the Soapy SDRPlay source and have the sdrplay service up and running. Also lsusb sees my RSP1A.
Whenever I try to execute my sdr file, there is always the run time error SoapySDR::Device::make() no match.
I am running UBUNTU 22.04.1 LTS with a fresh installation of radioconda.
Any idea on the missing link?
Regards

ryanvolz, 2022-10-17:
Hi @jmillo! Unfortunately SDRPlay is difficult because the source code needed to build in support in various packages (e.g. Soapy) is behind a license agreement and not automatically downloadable. That does not play well with the conda-forge approach for building packages using cloud CI resources. So currently radioconda does not come with the SoapySDR SDRPlay driver, even though GNU Radio provides a block that makes it look like it does. Your issue is that the Soapy library within the radioconda environment can't find that driver, even though you have it installed through Ubuntu outside that environment.

The proper fix is to figure out a way to get the Soapy SDRPlay driver packaged and included in radioconda. No guarantees on when that could happen, so until then you might be able to get by manually building SoapySDRPlay. Activate your radioconda environment, e.g. `conda activate base` if you don't have it set up to do that automatically. Then
```
git clone https://github.com/pothosware/SoapySDRPlay.git
cd SoapySDRPlay
mkdir build
cd build
cmake ..
make install
```
Then you will hopefully get positive results with `SoapySDRUtil -info` and `SoapySDRUtil --probe="driver=sdrplay"`, and it should work within the radioconda environment including in GNU Radio Companion.

jmillo, 2022-10-18:
Hi Ryan. Your solution works as a charm.
You just have to make sure that the resulting .so gets installed in the proper directory. In my case it is ~/miniconda3/envs/radioconda/lib/SoapySDR/modules0.8 as I did a miniconda install as per your video (https://www.youtube.com/watch?v=cWKecHoNo0Q).
Thanks again

closed 2022-10-18 by jmillo