radioconda/radioconda-installer #30: GNU Radio - DLL load failed while importing qtgui_python

ratzrattillo, 2022-10-17:
I am experiencing the following issue when trying to run a Flowgraph in GNU Radio with a QT GUI Frequency Sink.
Why is this happening?
The used operating System is Windows 11, Build 10.0.22621.674
I am using the latest build of radioconda: https://github.com/ryanvolz/radioconda/releases/tag/2022.09.22

```
Traceback (most recent call last):
  File "C:\Users\Username\radioconda\lib\site-packages\gnuradio\qtgui\__init__.py", line 19, in <module>
    from .qtgui_python import *
ImportError: DLL load failed while importing qtgui_python: The module could not be found
```

Why is this happening? Are dependencies missing?

ryanvolz, 2022-10-17:
Hmm. What does it say when you open a Conda Prompt and run the following commands?:
```
python -c "import sys; print('\n'.join(sys.path))"
python -c "from gnuradio.qtgui import qtgui_python"
```

ratzrattillo, 2022-10-18:
Thanks for the fast response! Here is the output of the commands:

```
(base) C:\Users\MyUser>python -c "import sys; print('\n'.join(sys.path))"

C:\Users\MyUser\radioconda\python310.zip
C:\Users\MyUser\radioconda\DLLs
C:\Users\MyUser\radioconda\lib
C:\Users\MyUser\radioconda
C:\Users\MyUser\radioconda\lib\site-packages

(base) C:\Users\MyUser>python -c "from gnuradio.qtgui import qtgui_python"
Traceback (most recent call last):
  File "C:\Users\MyUser\radioconda\lib\site-packages\gnuradio\blocks\__init__.py", line 18, in <module>
    from .blocks_python import *
ImportError: DLL load failed while importing blocks_python: The specified procedure could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\MyUser\radioconda\lib\site-packages\gnuradio\qtgui\__init__.py", line 49, in <module>
    from .auto_correlator_sink import AutoCorrelatorSink
  File "C:\Users\MyUser\radioconda\lib\site-packages\gnuradio\qtgui\auto_correlator_sink.py", line 17, in <module>
    from gnuradio import blocks, fft, filter
  File "C:\Users\MyUser\radioconda\lib\site-packages\gnuradio\blocks\__init__.py", line 22, in <module>
    from .blocks_python import *
ImportError: DLL load failed while importing blocks_python: The specified procedure could not be found
````

ryanvolz, 2022-10-18:
That's some progress in figuring out what's happening, but I still need more information about what DLL is failing to load. It's going to put out a lot of text, but could you report the output of the following (again from the Conda Prompt)?:
```
python -v -c "from gnuradio.qtgui import qtgui_python"
```

ratzrattillo, 2022-10-19:
Output:
```
(base) C:\Users\MyUser>python -v -c "from gnuradio.qtgui import qtgui_python" > out.txt
import _frozen_importlib # frozen
import _imp # builtin
import '_thread' # <class '_frozen_importlib.BuiltinImporter'>
import '_warnings' # <class '_frozen_importlib.BuiltinImporter'>
import '_weakref' # <class '_frozen_importlib.BuiltinImporter'>
import '_io' # <class '_frozen_importlib.BuiltinImporter'>
import 'marshal' # <class '_frozen_importlib.BuiltinImporter'>
import 'nt' # <class '_frozen_importlib.BuiltinImporter'>
import 'winreg' # <class '_frozen_importlib.BuiltinImporter'>
import '_frozen_importlib_external' # <class '_frozen_importlib.FrozenImporter'>
# installing zipimport hook
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
import 'zipimport' # <class '_frozen_importlib.FrozenImporter'>
# installed zipimport hook
# C:\Users\MyUser\radioconda\lib\encodings\__pycache__\__init__.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\encodings\__init__.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\encodings\\__pycache__\\__init__.cpython-310.pyc'
# C:\Users\MyUser\radioconda\lib\__pycache__\codecs.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\codecs.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\codecs.cpython-310.pyc'
import '_codecs' # <class '_frozen_importlib.BuiltinImporter'>
import 'codecs' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A259EE00>
# C:\Users\MyUser\radioconda\lib\encodings\__pycache__\aliases.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\encodings\aliases.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\encodings\\__pycache__\\aliases.cpython-310.pyc'
import 'encodings.aliases' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F43A0>
import 'encodings' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A259EC50>
# C:\Users\MyUser\radioconda\lib\encodings\__pycache__\utf_8.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\encodings\utf_8.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\encodings\\__pycache__\\utf_8.cpython-310.pyc'
import 'encodings.utf_8' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A259E9B0>
# C:\Users\MyUser\radioconda\lib\encodings\__pycache__\cp1252.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\encodings\cp1252.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\encodings\\__pycache__\\cp1252.cpython-310.pyc'
import 'encodings.cp1252' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F44F0>
import '_signal' # <class '_frozen_importlib.BuiltinImporter'>
# C:\Users\MyUser\radioconda\lib\__pycache__\io.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\io.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\io.cpython-310.pyc'
# C:\Users\MyUser\radioconda\lib\__pycache__\abc.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\abc.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\abc.cpython-310.pyc'
import '_abc' # <class '_frozen_importlib.BuiltinImporter'>
import 'abc' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F4970>
import 'io' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F4760>
# C:\Users\MyUser\radioconda\lib\__pycache__\site.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\site.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\site.cpython-310.pyc'
# C:\Users\MyUser\radioconda\lib\__pycache__\os.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\os.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\os.cpython-310.pyc'
# C:\Users\MyUser\radioconda\lib\__pycache__\stat.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\stat.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\stat.cpython-310.pyc'
import '_stat' # <class '_frozen_importlib.BuiltinImporter'>
import 'stat' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F6BC0>
# C:\Users\MyUser\radioconda\lib\__pycache__\_collections_abc.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\_collections_abc.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\_collections_abc.cpython-310.pyc'
import '_collections_abc' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F6EF0>
# C:\Users\MyUser\radioconda\lib\__pycache__\ntpath.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\ntpath.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\ntpath.cpython-310.pyc'
# C:\Users\MyUser\radioconda\lib\__pycache__\genericpath.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\genericpath.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\genericpath.cpython-310.pyc'
import 'genericpath' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A2639930>
import '_winapi' # <class '_frozen_importlib.BuiltinImporter'>
import 'ntpath' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F70A0>
import 'os' # <_frozen_importlib_external.SourceFileLoader object at 0x000001C1A25F57E0>
# C:\Users\MyUser\radioconda\lib\__pycache__\_sitebuiltins.cpython-310.pyc matches C:\Users\MyUser\radioconda\lib\_sitebuiltins.py
# code object from 'C:\\Users\\MyUser\\radioconda\\lib\\__pycache__\\_sitebuiltins.cpython-310.pyc'
[... cut ...]
# destroy numpy.lib.mixins
# destroy numpy.lib.ufunclike
# destroy numpy.lib.type_check
# destroy numpy.lib.stride_tricks
# destroy numpy.lib.twodim_base
# destroy numpy.lib.histograms
# destroy numpy.lib.index_tricks
# destroy numpy.lib.nanfunctions
# destroy numpy.lib.shape_base
# destroy numpy.lib.polynomial
# destroy numpy.lib.utils
# destroy numpy.lib.arraysetops
# destroy numpy.lib._datasource
# destroy numpy.lib._iotools
# destroy numpy.lib.npyio
# destroy numpy.lib.arrayterator
# destroy nump

ryanvolz, 2022-10-19:
There's lots there, but it still just fails at importing `blocks_python` without any more informative message.

Maybe try reinstalling gnuradio/radioconda?

ratzrattillo, 2022-10-21:
I tried to uninstall and reinstall, but unfortunately this did not work. I did not experience this issue on another Windows 10 System, but on Windows 11 with the build number 10.0.22621.674.
Do any specific PATH variables have to be set to specific values?

ryanvolz, 2022-10-21:
Yes, but the PATH should be set appropriately when you activate the conda environment. Still worth checking on the PATH from the Conda Prompt just in case.

It's interesting that this could be a Windows 11 thing. I'm not familiar enough with what the differences are for that to help me narrow down candidates, but it's something.

ratzrattillo, 2022-10-22:
Here the output of %PATH% in the Conda Prompt:

```
echo %PATH%

(base) C:\Users\MyUser>echo %PATH%
C:\Users\MyUser\radioconda;
\C:\Users\MyUser\radioconda\Library\mingw-w64\bin;
\C:\Users\MyUser\radioconda\Library\usr\bin;
\C:\Users\MyUser\radioconda\Library\bin;
\C:\Users\MyUser\radioconda\Scripts;
\C:\Users\MyUser\radioconda\bin;
\C:\Users\MyUser\radioconda\condabin;
\.;
\C:\Windows\system32;
\C:\Windows;
\C:\Windows\System32\Wbem;
\C:\Windows\System32\WindowsPowerShell\v1.0;
\C:\Windows\System32\OpenSSH;
\C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin;
\C:\Program Files\bladeRF\x64;
\C:\Program Files\CMake\bin;
\C:\WINDOWS\system32;
\C:\WINDOWS;
\C:\WINDOWS\System32\Wbem;
\C:\WINDOWS\System32\WindowsPowerShell\v1.0;
\C:\WINDOWS\System32\OpenSSH;
\C:\Program Files\LLVM\bin;
\C:\Program Files\dotnet;
\C:\Program Files\PothosSDR\lib;
\C:\Program Files\PothosSDR\bin;
\C:\Users\MyUser\AppData\Local\Microsoft\WindowsApps;
\C:\Users\MyUser\.dotnet\tools;
\.
```

ratzrattillo, 2022-10-22:
There seems to be a general issue in resolving names. See also the following image:

[https://ibb.co/17w5FPJ](url)

jmillo, 2022-10-27:
I seem to have the same type of problem.
Here are the files showing the error during install and upgrade dialog.
I tried reinstalling to no avail.
My system is Windows11 Pro Version 22H2, build 22621.674
[image]
[radiocondainstall.txt](https://github.com/ryanvolz/radioconda/files/9882757/radiocondainstall.txt)
There is definitely something fishy with GTK.

ryanvolz, 2022-10-27:
@jmillo What does your `%PATH%` look like? (`echo %PATH%` from conda prompt with activated radioconda environment)

I'm not sure if your issue is ultimately the same or not, but there are certainly perplexing things in your logs.

Also, your `conda upgrade --all` replaced a lot of packages with versions from the defaults channel, which might introduce incompatibilities with the conda-forge packages. I should probably improve the README there to avoid that conflict. But you probably want to run `conda upgrade --all -c conda-forge -c ryanvolz` to get back to known compatible packages.

jmillo, 2022-10-28:
Here is my path:

(base) C:\Users\Jean>conda activate radioconda

(radioconda) C:\Users\Jean>echo %PATH%
D:\Anaconda3\envs\radioconda;D:\Anaconda3\envs\radioconda\Library\mingw-w64\bin;D:\Anaconda3\envs\radioconda\Library\usr\bin;D:\Anaconda3\envs\radioconda\Library\bin;D:\Anaconda3\envs\radioconda\Scripts;D:\Anaconda3\envs\radioconda\bin;D:\Anaconda3\condabin;C:\Program Files\Microsoft MPI\Bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files\dotnet;C:\Program Files\Microsoft SQL Server\130\Tools\Binn;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn;C:\Program Files (x86)\Common Files\Acronis\VirtualFile;C:\Program Files (x86)\Common Files\Acronis\VirtualFile64;C:\Program Files (x86)\Common Files\Acronis\FileProtector;C:\Program Files (x86)\Common Files\Acronis\FileProtector64;C:\Program Files (x86)\Common Files\Acronis\SnapAPI;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn;C:\Program Files\Microsoft SQL Server\140\Tools\Binn;C:\Program Files (x86)\Microsoft SQL Server\140\DTS\Binn;C:\Program Files\Microsoft SQL Server\140\DTS\Binn;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn;C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn;D:\Program Files\PuTTY;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\IVI Foundation\VISA\Win64\Bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;D:\msys64\mingw64\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;D:\Program Files\Terminotix\LogiTerm\server\php;C:\Program Files\Microsoft SQL Server\150\Tools\Binn;C:\Program Files (x86)\dotnet;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\Git\cmd;C:\Users\Jean\.cargo\bin;C:\Users\Jean\AppData\Local\Microsoft\WindowsApps;C:\Users\Jean\.dotnet\tools;D:\Program Files\Graphviz\bin;D:\Program Files\Microsoft VS Code\bin;.

(radioconda) C:\Users\Jean>

------> new update command + companion test (same GTK problem)

(radioconda) C:\Users\Jean>conda upgrade --all -c conda-forge -c ryanvolz
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: D:\Anaconda3\envs\radioconda

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    gnuradio-satellites-5.1.0  |  py310h0159cc5_0         687 KB  conda-forge
    hackrf-2022.09.1           |       h9647d90_1          69 KB  conda-forge
    libhackrf0-2022.09.1       |       hcfcfb64_1          31 KB  conda-forge
    libxslt-1.1.37             |       h0192164_0         505 KB  conda-forge
    lxml-4.9.1                 |  py310hc0e5b84_1         1.0 MB  conda-forge
    matplotlib-3.6.1           |  py310h5588dad_0           8 KB  conda-forge
    pandas-1.5.1               |  py310h1c4a608_1        11.7 MB  conda-forge
    pillow-9.2.0               |  py310hd4fb230_3        45.3 MB  conda-forge
    soapysdr-0.8.1             |  py310h232114e_3         475 KB  conda-forge
    ------------------------------------------------------------
                                           Total:        59.8 MB

The following NEW packages will be INSTALLED:

  backports          conda-forge/noarch::backports-1.0-py_2 None
  backports.functoo~ conda-forge/noarch::backports.functools_lru_cache-1.6.4-pyhd8ed1ab_0 None
  cached_property    conda-forge/noarch::cached_property-1.5.2-pyha770c72_1 None
  contourpy          conda-forge/win-64::contourpy-1.0.5-py310h232114e_1 None
  fonts-conda-forge  conda-forge/noarch::fonts-conda-forge-1-0 None
  importlib-metadata conda-forge/noarch::importlib-metadata-5.0.0-pyha770c72_1 None
  importlib_resourc~ conda-forge/noarch::importlib_resources-5.10.0-pyhd8ed1ab_0 None
  lame               conda-forge/win-64::lame-3.100-hcfcfb64_1003 None
  lcms2              conda-forge/win-64::lcms2-2.12-h2a16943_0 None
  libblas            conda-forge/win-64::libblas-3.9.0-16_win64_mkl None
  libcblas           conda-forge/win-64::libcblas-3.9.0-16_win64_mkl None
  liblapack          conda-forge/win-64::liblapack-3.9.0-16_win64_mkl None
  libxcb             conda-forge/win-64::libxcb-1.13-hcd874cb_1004 None
  mpg123             conda-forge/win-64::mpg123-1.30.2-h63175ca_1 None
  openjpeg           conda-forge/win-64::openjpeg-2.5.0-hc9384bd_1 None
  pkgutil-resolve-n~ conda-forge/noarch::pkgutil-resolve-name-1.3.10-pyhd8ed1ab_0 None
  pthread-stubs      conda-forge/win-64::pthread-stubs-0.4-hcd874cb_1001 None
  tbb                conda-forge/win-64::tbb-2021.6.0-h91493d7_1 None
  typing_extensions  conda-forge/noarch::typing_extensions-4.4.0-pyha770c72_0 None
  unicodedata2       conda-forge/win-64::unicodedata2-14.0.0-py310h8d17308_2 None
  xorg-libxau        conda-forge/win-64::xorg-libxau-1.0.9-hcd874cb_0 None
  xorg-libxdmcp      conda-forge/win-64::xorg-libxdmcp-1.1.3-hcd874cb_0 None
  zipp               conda-forge/noarch::zipp-3.10.0-pyhd8ed1ab_0 None

The following packages will be REMOVED:

  blas-1.0-mkl
  bottleneck-1.3.5-py310h9128911_0
[... cut ...]
** (gnuradio-companion.exe:19568): WARNING **: 13:08:26.241: Failed to load shared library 'gtk-3-vs15.dll' referenced by the typelib: 'gtk-3-vs15.dll': The specified procedure could not be found.

** (gnuradio-companion.exe:19568): WARNING **: 13:08:26.273: Failed to load shared library 'atk-1.0-0.dll' referenced by the typelib: 'atk-1.0-0.dll': The specified procedure could not be found.
While trying to display an error message, another error occurred
must be an interface
The original error message follows.
TypeError

Failed to initialize GTK. If you are running over ssh, did you enable X forwarding and start ssh with -X?

(must be an interface)

(radioconda) C:\Users\Jean>

AWADALLA333, 2022-10-28:
Hello @ryanvolz ,
I am facing the same error as @jmillo above.
I'm on Windows 11 Pro 22H2 (Build 22621.755)
Using the Radioconda 2022.09.22 Installer.

[image]

While executing the commands in the Notepad (in the screenshot 1 above) inside the Conda Prompt, in that specific order, this error message pops up.

[image]

Later on, when running the gnuradio-companion command (screenshot 2), the same error as the message above shows up again, then the Conda Prompt complains from missing (gtk-3-vs15.dll) and (atk-1.0-0.dll).

I'm willing to test it further if you can any suggest solution.

jmillo, 2022-10-31:
I don't know if this can save time to somebody, but the only entry point in atk-1.0-0.dll that pertains to setlocale is:

295  126 0001D710 libintl_set_relocation_prefix

ryanvolz, 2022-11-01:
I think these recent errors with the GTK stack are different from the original qtgui error, so I've created a separate issue to track it (#37). I can reproduce the GTK errors with the latest conda-forge packages, but the most recent radioconda installer is fine, so it should be short work to track down what changed.

ryanvolz, 2022-11-03:
@ratzrattillo To get back to your issue, maybe we can use `dlltracer` to get more information. [You'll have to be an Administrator though](https://github.com/microsoft/dlltracer-python#use).

1. Install dlltracer: `mamba install dlltracer`
2. Start ipython: `ipython`
3. Run the following:
    ```
    import dlltracer
    import sys

    with dlltracer.Trace(out=sys.stdout):
        from gnuradio.blocks import blocks_python
    ```

ratzrattillo, 2022-11-08:
I tried it again, following also the install instructions at https://github.com/conda-forge/dlltracer-feedstock, but even the installation of dlltracer in the conda prompt fails (Started as administrator):
```
Looking for: ['dlltracer']

conda-forge/win-64                                          Using cache
conda-forge/noarch                                          Using cache
ryanvolz/win-64                                               No change
ryanvolz/noarch                                               No change

Pinned packages:
  - python 3.10.*

Encountered problems while solving:
  - nothing provides requested dlltracer

(base) C:\WINDOWS\system32>
```

Trying to import dlltracer in ipython leads to:
```
Python 3.10.5 | packaged by conda-forge | (main, Jun 14 2022, 06:57:19) [MSC v.1929 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import dlltracer
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 import dlltracer

ModuleNotFoundError: No module named 'dlltracer'

In [2]: exit
```

I also installed a previous version of radioconda and i have the same issue. Now also on a Windows 10 system. It is really weird. Is there anything that definitely must be in the PATH or definitely NOT in the PATH?
Are there any restrictions regarding the permissions with which radioconda has to be installed?

ryanvolz, 2022-11-08:
> 1. Install dlltracer: `mamba install -c "conda-forge/label/dlltracer_dev" dlltracer`

Ugh, they put dlltracer in a separate channel for some reason. I updated the command so it reads as above, and now I've actually tried it myself to verify it works.

I didn't run the "Conda Prompt" as administrator before installing dlltracer, but I did have to close it and reopen it as an Administrator so that step 3 would work.

> I also installed a previous version of radioconda and i have the same issue. Now also on a Windows 10 system. It is really weird. Is there anything that definitely must be in the PATH or definitely [...]

It could be something like this, but your PATH looks fine to me and I'm not aware of any permissions restrictions. If I had to guess, there's probably some other software that you're installing that has a competing version of some library which is causing problems. Hopefully the dlltrace will be informative.

ratzrattillo, 2022-11-09:
Now it worked:

```
In [11]: import dlltracer
    ...: import sys
    ...:
    ...: with dlltracer.Trace(out=sys.stdout):
    ...:     from gnuradio.blocks import blocks_python
    ...:
```

The import did not return any error.

I tried again another import from a different directory:

```
(base) C:\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\qtgui>ipython
Python 3.10.5 | packaged by conda-forge | (main, Jun 14 2022, 06:57:19) [MSC v.1929 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from gnuradio import qtgui
   ...: from .qtgui_python import *
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:18, in <module>
     17 try:
---> 18     from .blocks_python import *
     19 except ImportError:

ImportError: DLL load failed while importing blocks_python: The procedure was not found.

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 from gnuradio import qtgui
      2 from .qtgui_python import *

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\qtgui\__init__.py:49, in <module>
     47 from .toggleswitch import GrToggleSwitch
     48 from .graphicoverlay import GrGraphicOverlay
---> 49 from .auto_correlator_sink import AutoCorrelatorSink
     50 from .auto_correlator_sink import AutoCorrelator
     51 from .auto_correlator_sink import Normalize

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\qtgui\auto_correlator_sink.py:17, in <module>
     15 from gnuradio import gr
     16 from gnuradio import qtgui
---> 17 from gnuradio import blocks, fft, filter
     19 from PyQt5 import QtGui
     20 from PyQt5.QtWidgets import QWidget

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:22, in <module>
     20     dirname, filename = os.path.split(os.path.abspath(__file__))
     21     __path__.append(os.path.join(dirname, "bindings"))
---> 22     from .blocks_python import *
     24 from .sigmf_sink_minimal import sigmf_sink_minimal
     25 from .stream_to_vector_decimator import *

ImportError: DLL load failed while importing blocks_python: Procedure not found.

```

ratzrattillo, 2022-11-09:
I think i got the issue:

```
(base) C:\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\qtgui>ipython
Python 3.10.5 | packaged by conda-forge | (main, Jun 14 2022, 06:57:19) [MSC v.1929 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.4.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: from gnuradio import qtgui
   ...: from .qtgui_python import *
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:18, in <module>
     17 try:
---> 18     from .blocks_python import *
     19 except ImportError:

ImportError: DLL load failed while importing blocks_python: Procedure not found.

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 from gnuradio import qtgui
      2 from .qtgui_python import *

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\qtgui\__init__.py:49, in <module>
     47 from .toggleswitch import GrToggleSwitch
     48 from .graphicoverlay import GrGraphicOverlay
---> 49 from .auto_correlator_sink import AutoCorrelatorSink
     50 from .auto_correlator_sink import AutoCorrelator
     51 from .auto_correlator_sink import Normalize

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\qtgui\auto_correlator_sink.py:17, in <module>
     15 from gnuradio import gr
     16 from gnuradio import qtgui
---> 17 from gnuradio import blocks, fft, filter
     19 from PyQt5 import QtGui
     20 from PyQt5.QtWidgets import QWidget

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:22, in <module>
     20     dirname, filename = os.path.split(os.path.abspath(__file__))
     21     __path__.append(os.path.join(dirname, "bindings"))
---> 22     from .blocks_python import *
     24 from .sigmf_sink_minimal import sigmf_sink_minimal
     25 from .stream_to_vector_decimator import *

ImportError: DLL load failed while importing blocks_python: Procedure not found.

In [2]: import dlltracer
   ...: import sys
   ...:
   ...: with dlltracer.Trace(out=sys.stdout):
   ...:     from gnuradio.blocks import blocks_python
   ...:
LoadLibrary \Device\HarddiskVolume6\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\blocks\blocks_python.cp310-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Program Files\PothosSDR\bin\gnuradio-blocks.dll
Failed \Device\HarddiskVolume6\Program Files\PothosSDR\bin\gnuradio-blocks.dll
Failed \Device\HarddiskVolume6\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\blocks\blocks_python.cp310-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\blocks\blocks_python.cp310-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Program Files\PothosSDR\bin\gnuradio-blocks.dll
Failed \Device\HarddiskVolume6\Program Files\PothosSDR\bin\gnuradio-blocks.dll
Failed \Device\HarddiskVolume6\Users\MyUser\AppData\Local\radioconda\Lib\site-packages\gnuradio\blocks\blocks_python.cp310-win_amd64.pyd
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:18, in <module>
     17 try:
---> 18     from .blocks_python import *
     19 except ImportError:

ImportError: DLL load failed while importing blocks_python: Procedure not found.

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
Input In [2], in <cell line: 4>()
      2 import sys
      4 with dlltracer.Trace(out=sys.stdout):
----> 5     from gnuradio.blocks import blocks_python

File ~\AppData\Local\radioconda\lib\site-packages\gnuradio\blocks\__init__.py:22, in <module>
     20     dirname, filename = os.path.split(os.path.abspath(__file__))
     21     __path__.append(os.path.join(dirname, "bindings"))
---> 22     from .blocks_python import *
     24 from .sigmf_sink_minimal import sigmf_sink_minimal
     25 from .stream_to_vector_decimator import *

ImportError: DLL load failed while importing blocks_python: Procedure not found.
```

Seems like it is connected with a parallel install of PhotosSDR.

ryanvolz, 2022-11-09:
> Uninstalling PhotosSDR solved the issue for me. The root cause is probably that both tools exposed the same DLLs in the PATH.
> 
> I think this issue could be closed from my side. Thanks for all the support. Maybe there could be a check implemented, to see if anything interferes in the PATH?

Excellent, that narrows it down nicely. I'll have to try to figure out what mechanism PothosSDR is employing to make its DLLs have priority and come up with a way to prevent it, because I don't want the only answer to be that you can only choose one of the environments.

I'll leave this open as a reminder to do that.

ryanvolz, 2023-04-12:
Mitigations for interference from PothosSDR have been in the last few releases, and I think that has improved things. Feel free to re-open if that's not the case.

closed 2023-04-12 by ryanvolz

jvoeller0129, 2024-01-11:
> Uninstalling PhotosSDR solved the issue for me. The root cause is probably that both tools exposed the same DLLs in the PATH.
> 
> I think this issue could be closed from my side. Thanks for all the support. Maybe there could be a check implemented, to see if anything interferes in the PATH?

this worked for me also thanks :)

PedroA-Vieira, 2024-02-09:
Just for the record...
I had this same problem:
"ImportError: DLL load failed while importing qtgui_python: The module could not be found"
Resolved by uninstalling PothosSDR, restarting the computer and running gnuradio-companion in the created virtual environment. The installation I did was using Conda, as follows:
https://wiki.gnuradio.org/index.php?title=CondaInstall