radioconda/radioconda-installer #31: Embedded Python blocks won't run on M1 Mac with Arm64 build

techfoamer, 2022-10-28:
Simple flowgraph with just the default embedded Python block quits when trying to run.

[problem.txt](https://github.com/ryanvolz/radioconda/files/9891109/problem.txt)

[python_block_test.pdf](https://github.com/ryanvolz/radioconda/files/9891131/python_block_test.pdf)

ryanvolz, 2022-10-28:
Ha! Looks like https://github.com/gnuradio/gnuradio/issues/6283, which I happen to be investigating right this moment!

This is a tough one. While not strictly related to conda, that seems to be where it is felt most painfully since conda-forge statically links the Python library into their Python interpreter and that's not very common.

ryanvolz, 2023-01-19:
The fix for this is in today's release!

closed 2023-01-19 by ryanvolz