radioconda/radioconda-installer #32: TX flowgraph makes a complaint about USB on M1 Mac with Arm64 build

techfoamer, 2022-10-28:
I'm using a USB Ettus B200mini.  Flowgraph runs the first time after launching GRC.  Subsequent launches randomly don't run, and say this:

[INFO] [UHD] Mac OS; Clang version 14.0.4 ; Boost_107800; UHD_4.3.0.0-release
Assertion failed: (list_empty(&darwin_cached_devices)), function darwin_init, file os/darwin_usb.c, line 605.

Flowgraph will usually run after closing GRC and unplugging then replugging the B200mini.

[usb_problem.txt](https://github.com/ryanvolz/radioconda/files/9891251/usb_problem.txt)

ryanvolz, 2022-10-28:
It looks like it crashes in a thread that is loading the `libbladeRF` library. Curious since you're using UHD. You could try uninstalling `soapysdr-module-bladerf` to see if that is truly causing the problem:
```
conda uninstall soapysdr-module-bladerf
```

techfoamer, 2022-10-28:
Hmm, lots of complaints after issuing the uninstall command.

[uninstall_soapysdr-module-bladerf_output.txt](https://github.com/ryanvolz/radioconda/files/9891403/uninstall_soapysdr-module-bladerf_output.txt)

ryanvolz, 2022-10-28:
Maybe `mamba uninstall soapysdr-module-bladerf` will work or give clearer output? It's weird, because nothing should depend on `soapysdr-module-bladerf`.

techfoamer, 2022-10-29:
That worked.  Flowgraphs with B200mini run without errors.

ryanvolz, 2023-01-20:
Did the new BladeRF release fix this, or do you still need the workaround of uninstalling `soapysdr-module-bladerf`?