radioconda/radioconda-installer #37: gtk stack broken on Windows with most recent conda-forge packages

ryanvolz, 2022-11-01:
Hello @ryanvolz ,
I am facing the same error as @jmillo above.
I'm on Windows 11 Pro 22H2 (Build 22621.755)
Using the Radioconda 2022.09.22 Installer.

[image]

While executing the commands in the Notepad (in the screenshot 1 above) inside the Conda Prompt, in that specific order, this error message pops up.

[image]

Later on, when running the gnuradio-companion command (screenshot 2), the same error as the message above shows up again, then the Conda Prompt complains from missing (gtk-3-vs15.dll) and (atk-1.0-0.dll).

I'm willing to test it further if you can any suggest solution.

_Originally posted by @AWADALLA333 in https://github.com/ryanvolz/radioconda/issues/30#issuecomment-1295489562_

ryanvolz, 2022-11-01:
The problem was introduced somewhere in this set of changed packages:
```
  Upgrade:
-----------------------------------------------------------------------------

  - fontconfig         2.14.0  hce3cb01_0     conda-forge
  + fontconfig         2.14.1  hbde0cde_0     conda-forge/win-64      213kB
  - gettext          0.19.8.1  ha2e2712_1008  conda-forge
  + gettext            0.21.1  h5728263_0     conda-forge/win-64        6MB
  - glib               2.72.1  h7755175_0     conda-forge
  + glib               2.74.1  h12be248_0     conda-forge/win-64      509kB
  - glib-tools         2.72.1  h7755175_0     conda-forge
  + glib-tools         2.74.1  h12be248_0     conda-forge/win-64      190kB
  - libgirepository    1.72.0  h234efe0_1     conda-forge
  + libgirepository    1.74.0  h4a5d6e9_1     conda-forge/win-64      278kB
  - libglib            2.72.1  h3be07f2_0     conda-forge
  + libglib            2.74.1  h79619a9_0     conda-forge/win-64        3MB
  - libiconv             1.16  he774522_0     conda-forge
  + libiconv             1.17  h8ffe710_0     conda-forge/win-64      715kB
  - libzlib            1.2.12  hcfcfb64_3     conda-forge
  + libzlib            1.2.13  hcfcfb64_4     conda-forge/win-64       71kB
  - zlib               1.2.12  hcfcfb64_3     conda-forge
  + zlib               1.2.13  hcfcfb64_4     conda-forge/win-64      116kB
```

ryanvolz, 2022-11-01:
`glib` is the problem, downgrading to `2.74.0` fixes things for me:
```
  Package       Version  Build       Channel                 Size
-------------------------------------------------------------------
  Downgrade:
-------------------------------------------------------------------

  - glib         2.74.1  h12be248_0  conda-forge
  + glib         2.74.0  h12be248_0  conda-forge/win-64     463kB
  - glib-tools   2.74.1  h12be248_0  conda-forge
  + glib-tools   2.74.0  h12be248_0  conda-forge/win-64     172kB
  - libglib      2.74.1  h79619a9_0  conda-forge
  + libglib      2.74.0  h79619a9_0  conda-forge/win-64       3MB
```

jmillo, 2022-11-01:
Yes this definitely seems to solve the problem.
I just used the following line:
     conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda

Thanks for your prompt reply.

AWADALLA333, 2022-11-01:
Yes. It works now.
I confirm that downgrading glib to 2.74.0 fixes this issue.

`-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-/\-`

Documenting the **temporary** [until the broken _gtk_ lib is fixed on Win11] fix I followed _for whoever googling this issue later_:

I ran these commands in a fresh (_Radioconda 2022.09.22 Installer_) Conda Promt:

1. `conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda`
2. `conda create -n gnuradio`

3. Then downgrading glib (3 packages [`glib, glib-tools, libglib`] will be downgraded 'together' through this command):
`conda install libglib=2.74.0`

4. Then proceeded to "pin" (disabling the update of glib packages to the recent version 2.74.1 ), following this guide: ( https://docs.conda.io/projects/conda/en/latest/user-guide/tasks/manage-pkgs.html#preventing-packages-from-updating-pinning )
       a. Create an extension-less file named (pinned) in the folder: `C:\Users\<USER_PROFILE>\radioconda\envs\gnuradio\conda-meta`
       b. Edit the file (pinned) using Notepad.
       c. Add these lines:
```
glib=2.74.0
glib-tools=2.74.0
libglib=2.74.0
```
d. Save and proceed.

5. Installing and running gnuradio in the set environment:

```
conda install gnuradio
conda activate gnuradio
gnuradio-companion
```

It should work now.
Many thanks @jmillo :wink:
Much appreciated @ryanvolz :smiley:

jmillo, 2022-11-02:
Hi to all,
I goofed in my comment the following line does it in a single stroke (I copied the wrong one):
conda create -n radioconda -c conda-forge -c ryanvolz --only-deps libglib=2.74.0 radioconda
Regards

ryanvolz, 2022-11-02:
This should be fixed now after https://github.com/conda-forge/atk-feedstock/pull/21 and an upgrade of the `atk` and `atk-1.0` packages to the latest versions, which allows using the latest `glib` 2.74.1 packages. Let me know if that's not the case.

closed 2022-11-02 by ryanvolz