radioconda/radioconda-installer #38: SURVEY: What's your experience with radioconda? What bundled packages do you use? What packages should be added next?

ei8fdb, 2022-11-03:
This is an invitation to all radioconda users to take part in.

## What is this survey about?

This survey is to learn what people use Radioconda for, what radio software packages bundled with it they use most, and what other software packages users would like to see included in the future. It has 6 short questions.

If you know of people using radioconda, you're welcome to send the survey link onto them.

https://saneuxdesign.survey.fm/radioconda-survey

## Getting involved in radioconda packaging

If you're interested in helping package new software to be included in Radioconda, you can also leave us your email address and we'll get in touch.

## What will the data collected be used for?

It will be used to understand how to Radioconda is used, what packaged to include in the future. Nothing else.

## Who created this survey?

It was created by Bernard Tyers, you can contact him [by email](mailto:bernard+radio@ei8fdb.org).

(This is obviously not an issue - I'm posting this here as many radioconda users come to the repo for support. I've asked Ryan if it's OK to post this here and he's agreed.)

csylvain, 2025-03-23:
please add `sounddevice` python package (wrapper for <ins>portaudio</ins>) - provides simple means of troubleshooting audio device selection.

with package installed, the command `python -m sounddevice` in the conda environment produces an enumeration and names of audio devices available to the <ins>portaudio</ins> backend.

see https://github.com/gnuradio/gnuradio/issues/7783 for additional information

csylvain, 2025-08-07:
please add `gnuradio-hpsdr` for platform `win-64` (exists already for `linux-64`)