radioconda/radioconda-installer #40: "Failed to initialize GTK." running grc after "conda upgrade --all"

DeonMarais64, 2022-11-09:
Hi,
On a Windows 11 machine I am getting the following errors,
After a fresh install of the latest release (2022.09.22) an then running `conda upgrade --all`

[image]

Any ideas on how to solve this if indeed possible ?
The only way I can get things working is to remove and and reinstall `radioconda`.

Also while I am here, a couple of other things.
In a windows `command` prompt,

```
C:\Users\deon>iio_info -s
Library version: 0.24 (git tag: c4498c2)
Compiled with backends: xml ip usb serial
Available contexts:
        0: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9361)), serial=10000023135756120041002703111138a7 [ip:pluto.local]
        1: 192.168.1.40 (Analog Devices PlutoSDR Rev.C (Z7010-AD9361)), serial=1000002313575612002600260311112a3e [ip:plutoplus.local]
        2: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10000023135756120041002703111138a7 [usb:1.3.5]
```

On the same machine and in a `conda` prompt,
```
(base) C:\Users\deon>iio_info -s
Library version: 0.24 (git tag: v0.24)
Compiled with backends: xml ip usb
Unable to create Local IIO context : Function not implemented (40)
Available contexts:
        0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=10000023135756120041002703111138a7 [usb:2.2.5]
```
It seems the `conda` environment does not see devices that are network attached. Within `GRC` it is still possible to use the devices.
For example `ip:plutoplu.local` works within a `GRC` flow diagram and so does running the resulting python script on its own.
This is not really a showstopper, but it would be great if it could be resolved.

Can `SDRplay` support be added either retrospectively, or could it also be included as part of the installation ?

Lastly, part of the roadmap to include `gr-fosphor` in as part of the installation, if not could it be and is there any current method by which to install this and OOT GNURadio modules on Windows ?

Thanks and best regards.

ryanvolz, 2022-11-09:
> Any ideas on how to solve this if indeed possible ?

This error is similar to #37 and may have the same cause. Can you post the output of `conda list` after the upgrade? Either you're missing the newly-upgraded `atk` package, or other packages have the same error and seemingly need to be rebuilt.

> It seems the `conda` environment does not see devices that are network attached. Within `GRC` it is still possible to use the devices.
> For example `ip:plutoplus.local` works within a `GRC` flow diagram and so does running the resulting python script on its own.
> This is not really a showstopper, but it would be great if it could be resolved.

Interesting. I'll have to take a look with my Pluto to see if there's anything that can be done.

> Can `SDRplay` support be added either retrospectively, or could it also be included as part of the installation ?

I'd love to include it if it can be made possible. See #29.

> Lastly, part of the roadmap to include `gr-fosphor` in as part of the installation, if not could it be and is there any current method by which to install this and OOT GNURadio modules on Windows ?

`gr-fosphor` is on my personal priority list, but I'm unsure how difficult it may be. [This](https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio) is the general procedure to follow for building an OOT with conda + Windows, although `gr-fosphor` could be less straightforward than the general case. I'd be interested to hear if you are successful though or what problems you run into.

DeonMarais64, 2022-11-10:
Thank you for your response.

> Can you post the output of `conda list` after the upgrade?

I below is the list *before* the upgrade,
```
(base) C:\Users\deon>conda list
# packages in environment at C:\Programs\Radio\GNURadio\radioconda:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        41.0                 h57928b3_1    conda-forge
airspy                    1.0.10               h8ffe710_0    conda-forge
airspyhf                  1.6.8                h8ffe710_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
asttokens                 2.0.8              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hdff456e_0    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.2           py310he2412df_0    conda-forge
bladerf                   2021.10              h8ffe710_0    conda-forge
boost-cpp                 1.78.0               h9f4b32c_1    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
brotlipy                  0.7.0           py310he2412df_1004    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2022.9.14            h5b45459_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
certifi                   2022.9.14          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310hcbf9ad4_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
click                     8.1.3           py310h5588dad_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
codec2                    1.0.3                hcd874cb_0    conda-forge
colorama                  0.4.5              pyhd8ed1ab_0    conda-forge
conda                     4.14.0          py310h5588dad_0    conda-forge
conda-package-handling    1.9.0           py310h635b8f1_0    conda-forge
construct                 2.10.68            pyhd8ed1ab_0    conda-forge
contourpy                 1.0.5           py310h232114e_0    conda-forge
cryptography              37.0.4          py310ha857299_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
digital_rf                2.6.7           py310h3aa5818_3    conda-forge
epoxy                     1.5.10               h8d14728_1    conda-forge
executing                 1.0.0              pyhd8ed1ab_0    conda-forge
expat                     2.4.9                h1537add_0    conda-forge
fftw                      3.3.10          nompi_h38027f0_105    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hce3cb01_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.37.3          py310h8d17308_0    conda-forge
freetype                  2.12.1               h546665d_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
fs                        2.4.15             pyhd8ed1ab_0    conda-forge
gdk-pixbuf                2.42.8               h56bf1b2_1    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glib                      2.72.1               h7755175_0    conda-forge
glib-tools                2.72.1               h7755175_0    conda-forge
gnuradio                  3.10.4.0        py310h6c50af4_2    conda-forge
gnuradio-core             3.10.4.0        py310hf7df68c_2    conda-forge
gnuradio-grc              3.10.4.0        py310h4a85852_2    conda-forge
gnuradio-iio              3.10.4.0        py310hfd4675a_2    conda-forge
[... 444 lines cut ...]
urllib3                   1.26.12         py310haa95532_0
vc                        14.2                 h21ff451_1
volk                      2.5.2                h63175ca_1    conda-forge
vs2015_runtime            14.32.31332          h1d6e394_9    conda-forge
watchdog                  2.1.6           py310haa95532_0
wcwidth                   0.2.5              pyhd3eb1b0_0
wheel                     0.37.1             pyhd3eb1b0_0
win_inet_pton             1.1.0           py310haa95532_0
wincertstore              0.2             py310haa95532_2
wxwidgets                 3.2.1                hd1f7ea3_0    conda-forge
xz                        5.2.6                h8cc25b3_0
yaml                      0.2.5                he774522_0
[... cut ...]
I don't know if it could apply here but `cemaxecuter` in `DragonOS` provides preconfigured Soapy drivers and then once the OS is installed you need to run a script to install the SDRplay API were you need to agree to their license conditions. Could it possibly be done in a silimar fashion ?

> [This](https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio) is the general procedure to follow for building an OOT with conda + Windows

Thanks, I will have a go at that when I get some time to do so.

DeonMarais64, 2022-11-10:
I subsequently did another re-installation and then upgraded,

I did `mamba upgrade --all` as apposed to `conda upgrade --all`. 

I was under the impression that they effectively do the same thing ?

This time the upgrade went without any warnings or errors and `gqrx` now runs, however running `grc` gives the following error,
 
[image]

Thanks in advance

ryanvolz, 2022-11-10:
Curious that when I do `mamba upgrade --all` from my default install of 2022.09.22, it doesn't give me the same package upgrades. 

> I did `mamba upgrade --all` as apposed to `conda upgrade --all`.
> 
> I was under the impression that they effectively do the same thing ?

Maybe it's because `conda` and `mamba` don't actually give the same solution, although in principle they should.

Anyway, one thing I see from your upgraded package list is that many of the packages are now from the default (Anaconda, no channel name listed) channel instead of the conda-forge channel. I don't know how that happened, because the default channel should be disabled by, well, default. But the upshot is that there could definitely be binary incompatibility there of the type that you are seeing. Given that lots of things are throwing errors that are different, that's my guess. You also don't have the `atk 2.38.0 *_0` with `glib 2.74.1` combination where I previously saw the particular `intl.dll` error before, so I guess that's not it.

I would try setting
```
conda config --add channels conda-forge
conda config --set channel_priority strict
```
before another `mamba upgrade --all` to see if you can get a consistent conda-forge-only environment. I've just tweaked the configuration so that the next installer will set strict priority by default to try to avoid this problem.

Barring that, is there a reason you want to upgrade from the 2022.09.22 release? Might be easiest to stick with that, where at least I'm reasonably sure that set of packages works well together.

> I don't know if it could apply here but `cemaxecuter` in `DragonOS` provides preconfigured Soapy drivers and then once the OS is installed you need to run a script to install the SDRplay API were you need to agree to their license conditions. Could it possibly be done in a silimar fashion ?

I'll have to think about that. I'm pretty sure I've heard of a conda package or two that does something similar, so maybe I'll try to dig that information up.

DeonMarais64, 2022-11-11:
> I would try setting
```
conda config --add channels conda-forge
conda config --set channel_priority strict
```
before another ```mamba upgrade --all```

I did that after reinstalling and that seems to have solved the problem. So far everything works.

> Barring that, is there a reason you want to upgrade from the 2022.09.22 release? Might be easiest to stick with that, where at least I'm reasonably sure that set of packages works well together.

No reason in particular, other than having the impression that it is generally good practice to do so. I have hitherto  applied updates immediately after installation (which AFAIK is general and widespread advice and practice after most software packages are installed) and then every moth or so beyond that. I will henceforth be more circumspect about doing so. 

Thanks again for your responses. 

PS, I am unsure if I should close this issue. For that reason I will for the time being leave it open. Please indicate if you prefer that I do.

ryanvolz, 2022-11-11:
> No reason in particular, other than having the impression that it is generally good practice to do so. I have hitherto applied updates immediately after installation (which AFAIK is general and [...]

It's generally not a terrible idea, but evidently it has some risks! It's been illuminating for me because I didn't think that much binary incompatibility was lurking out there with the defaults channel, and that's resulted in me making a change that should help all users generally in the future. Thanks!

I think I'll close this since the problem has been figured out, but feel free to open new issues if you want to separately document the requests for gr-fosphor and SDRPlay. Those aren't likely to ever drop off my radar, but extra reminders couldn't hurt.

closed 2022-11-11 by ryanvolz