radioconda/radioconda-installer #42: Update with gr-gpredict-doppler

viniinpe, 2022-11-16:
Hello. I would like to know how to install this package on Gnuradio.
https://github.com/ghostop14/gr-gpredict-doppler

It's possible? Do you have a tutorial teaching?
Thank you very much in advance.

ryanvolz, 2022-11-17:
Looks very do-able. I'll add it to my list, along with `gpredict` if that makes sense.

In the meantime, this procedure should work pretty well for you: https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio

ryanvolz, 2023-03-09:
`gnuradio-gpredict-doppler` is now up on my conda-forge channel, installable with `mamba install -c ryanvolz gnuradio-gpredict-doppler`. I'll wait until this is in a radioconda release to close the issue.

viniinpe, 2023-03-13:
Thanks Daniel. It helped a lot, now I'm able to run the doppler correction on windows. Congratulations for your work. 
A doubt. Is it possible to get doppler correction from orbitron instead of Gpredict? Is there already a project developed or under development? Orbitron is better, as it allows you to send the frequency directly, without having to click on the engage button in Gpredict.

ryanvolz, 2023-03-14:
I'm not aware of any software that interfaces Orbitron and GNU Radio like gr-gpredict-doppler does for Gpredict.

closed 2023-04-12 by ryanvolz