radioconda/radioconda-installer #43: Need a way to point to an 'external' UHD installation or specify an alternative version

zbrozek, 2023-01-02:
Ettus dropped support for a series of Ethernet-connected radio boards with the UHD 4.x series, so I need to stay behind on UHD 3.15.0. I have it installed on my system, but radioconda installs its own and points other applications to it. Is there a way to override or replace the version?

ryanvolz, 2023-01-11:
There's no good way to point to an external UHD installation, because the main issue is that the various packages that depend on UHD would need to be compiled for any specific UHD version that you would want to use.

However, it would be reasonable to build all of the packages against both UHD 4 and UHD 3, and then the user could install the UHD version they need and it would pull in the compatible packages. It's a bit more work in maintaining the extra packages, so I'll have to look at how much more work it would actually be and decide if I'm up for it. It's good to know that there are use cases for sticking to UHD 3 (which boards did they drop support for?).

zbrozek, 2023-01-11:
That's probably more work than it's worth. The N230 was dropped, but also barely exists in the wild. I got one by buying a workbench at an auction for a failed startup, and it was in the drawer.