radioconda/radioconda-installer #44: Gnuradio Companion crashes on macOS Big Sur (M1)

Vinz87, 2023-01-11:
Hi, I installed radioconda on my Macbook Pro M1 with macOS Big Sur, and it seemed all fine because Gnuradio Companion opens correctly, but after adding some blocks as soon as I click on input/output ports to link blocks together the app crashes:

```
(base) Vincenzo@Vins-Macbook-Pro ~ % gnuradio-companion
>>> Warning: vocoder_codec2_decode_ps - option_attributes are for enums only, ignoring
>>> Warning: vocoder_codec2_encode_sp - option_attributes are for enums only, ignoring
<<< Welcome to GNU Radio Companion 3.10.4.0 >>>

Block paths:
	/Users/Vincenzo/radioconda/share/gnuradio/grc/blocks
zsh: segmentation fault  gnuradio-companion
```

ryanvolz, 2023-01-11:
There were some crashing bugs fixed recently that haven't quite made it into a radioconda release yet, so I suspect that's the issue. Try `mamba upgrade gnuradio` or `mamba upgrade --all` to see if you can get the most recent build of `gnuradio` 3.10.4.0 or `gnuradio` 3.10.50, which have the latest fixes. If you're running one of those and this is still an issue, then it's probably something we need to report upstream.

Vinz87, 2023-01-11:
Upgraded to gnuradio 3.10.5.0 and it works now!

closed 2023-01-11 by Vinz87