radioconda/radioconda-installer #45: radioconda-2023.01.19-Windows-x86_64 install "Failed to create menu" error

Phitchco, 2023-01-20:
trying for a fresh install of radioconda-2023.01.19-Windows-x86_64.exe for user only runs until subject error window appears.  If "Ignore" is chosen, then package completes.  The program shows up in the windows installed programs list, but there are no menu items.  Install details:

[radioconda install details.txt](https://github.com/ryanvolz/radioconda/files/10467207/radioconda.install.details.txt)

ryanvolz, 2023-01-20:
Thanks for the report! It looks like something in the underlying installer tool has changed to break this, either `constructor` or `conda` itself. The installer is running this command
```
"$INSTDIR\_conda.exe" constructor --prefix "$INSTDIR" --make-menus @MENU_PKGS@
```

and now getting this result:
```
The following argument was not expected: --make-menus
```

Hopefully a little digging will turn up what the change was and how to handle it.

ryanvolz, 2023-01-20:
Ah, not a `constructor` or `conda` change but a change I made after the last release to bundle `micromamba` instead of `conda-standalone`, resulting in: https://github.com/mamba-org/mamba/issues/1215. Looks like I'll have to roll back that change for the time being on Windows.

ryanvolz, 2023-01-21:
I just replaced the windows installer with a new build that successfully installed menu items for me. Let me know if it doesn't work for you.

closed 2023-01-21 by ryanvolz

Phitchco, 2023-01-21:
That did it.  installed with no issues and I now have GNU on my windows laptop.
thank you