radioconda/radioconda-installer #46: GRC with error "QObject::setParent: Cannot set parent, new parent is in a different thread"

Biran1911, 2023-01-29:
Hello,

I have used GRC for many applications on my desktop computer.
Just a few days ago I installed radioconda on my laptop (same way i did on my desktop), and now i get the above error whenever i use the QT GUI.

I have tried to reinstall, and install the package itself with no success, any suggestions?

print screen attached:
[image]

ryanvolz, 2023-01-30:
Does the QT GUI Time Sink give the same error? There have been [issues on Windows](#25) with anything that uses an FFT because of an issue with `fftw`, but this doesn't appear to be that.

It feels more like a GNU Radio bug than anything specific to conda packaging, so I think that's the place to look first. I'll try to run some tests locally to see if I can reproduce this.

ryanvolz, 2023-01-30:
I was not able to reproduce this, so the only thing I can think of is that you might have another Qt installation that is interfering with the radioconda one (although that shouldn't happen). Can you inspect your `%PATH%` and see if it points to anything before the radioconda directories that would have another Qt?

Biran1911, 2023-01-31:
> Does the QT GUI Time Sink give the same error? There have been [issues on Windows](https://github.com/ryanvolz/radioconda/issues/25) with anything that uses an FFT because of an issue with `fftw`, [...]

Indeed it does, If am sure it has to do with the conda package although I cannot seem to fix it (tried reinstall)

Biran1911, 2023-01-31:
> 

> %PATH%

path seems right, points to radioconda\Library\share\gnuradio\grc\blocks

atg1966, 2023-04-10:
I was having this exact same problem. My solution: removed the PyQt5 package from Thonny IDE.

gozillah, 2024-05-31:
I am having the same problem under Windows 10 since release 2023.11.06 and including 2024.05.29. I am still working with release 23.7.26. The  following elementary example runs fine with the recent releases, but without showing the time or the frequency sink:

[image]

The tone is audible and the frequency range widget is responsive, changing the pitch. Time or frequency sink are not visible. Activating either sink (time sink in the above snapshot) makes no difference. The problem is occuring with a freshly installed Radioconda without any OOT modules of mine. GR_PREFIX and GR_BLOCKS_PATH are set accurately. 

I am not using Thonny IDE, so I cannot apply the same fix as atg1966. 

I asked ChatGPT 3.5 for help

> In the gnuradio companion i get the following error message: "QObject::setParent: Cannot set parent, new parent is in a different thread". What may be the cause and where do i have to look for remedy?

and got a quite elaborate analysis with a recommendation to deep-dive into GNU Radio code, which is beyond my abilities.

ryanvolz, 2024-06-10:
So now that #78 is fixed, it will hopefully be easier to narrow in on this bug and get it fixed since it is clearly still an issue.

@gozillah You've separately mentioned a potential fix that involves setting the `PYTHONPATH` variable to `C:\Users\???\radioconda\Lib\site-packages`. That does solve your issue here? And it gets rid of the "QObject::setParent: Cannot set parent, new parent is in a different thread" message?

ryanvolz, 2024-06-11:
Putting this response here so I don't lose track of it. Interesting that the email reply didn't do that already...

> Correct. Deleting or omitting the Windows PYTHONPATH variable and then starting 6.11.2023, 26.1.2024 or 29.5.2024 Radioconda's Conda prompt, gnuradio-companion.exe and a flowgraph as e.g. FM Receiver creates the following results:
>
> -     The flowgraph is running (e.g. I do hear FM reception), 
> -     QT GUI widgets (e.g. QT GUI Range) are displayed and do work,
> -     but QT GUI graphic windows (e.g. Frequency Sink) are not displayed 
> -     and the companion's console displays the message "QObject::setParent: Cannot set parent, new parent is in a different thread".
>
> Setting the correct Windows PYTHONPATH variable (in my case C:\Users\my_windows_account\radioconda\Lib\site-packages), then starting the Conda prompt, gnuradio-companion.exe and the FM flowgraph makes everything run as expected and the above message is not displayed.

ost-bravo, 2024-07-01:
> Putting this response here so I don't lose track of it. Interesting that the email reply didn't do that already...

worked for me!