radioconda/radioconda-installer #47: pagesize :error: no info; setting pagesize = 4096

yourarchipelago, 2023-01-31:
Hi, 
This instruction always bumps up when running a grc or *.py file. 
It only appears on windows.

Just wanna know why and how to deal with it.

Thanks!

ryanvolz, 2023-01-31:
It's coming from [here](https://github.com/gnuradio/gnuradio/blob/25ad39f7a1a788c36e2ef712ac17b8f4b4d9891d/gnuradio-runtime/lib/pagesize.cc#L43). From the looks of it, there's currently no way to avoid this on Windows since both `getpagesize` and `sysconf` are only defined for Unix-like systems. This should probably be fixed in upstream GNU Radio to get the page size from a Windows-specific method. Evidently 4096 is a good enough default that is hasn't caused problems thus far, other than this overzealous error message.

ryanvolz, 2023-01-31:
I reported this upstream to determine the best course of action.

yourarchipelago, 2023-02-01:
The issue really hasn't caused any problem till now, hope that gnuradio could be more compatible with Windows in this year.
Thanks!

closed 2023-02-01 by yourarchipelago