radioconda/radioconda-installer #48: Сrashes when adding any QT module

Heorenmaru, 2023-02-06:
Windows10 1809 LTSC
Installation Folder: C:\SDR\radioconda

ryanvolz, 2023-02-06:
Hi @Heorenmaru! What do you mean by "adding any QT module"? Does it crash when you add a QT block to the flowgraph canvas? Or does running the flowgraph crash whenever a QT block is in use?  If this last one and it is specific to blocks involving an FFT, you might be seeing https://github.com/ryanvolz/radioconda/issues/25?

Heorenmaru, 2023-02-23:
[image]

ryanvolz, 2023-02-24:
Curious. I'm not able to reproduce the crash. I think the key question is: "Why QT GUI blocks and not others?". Do you also have Qt installed outside of radioconda, or some other programs or SDR software that use Qt and therefore installed it? We might have to elevate this one up to the GNU Radio issue tracker to see if they can offer insight.

Tobi82m, 2023-06-29:
I think i have the same issue. I have a windows notebook without admin rights. Today i have installed radioconda. There was no qt on the notebook bevor. For example, if i use the "qt gui range" and drop it into the window, gnuradio crash and close the window. Is there any solution for this problem? Reinstall of qtgui not solved the problem

Tobi82m, 2023-07-17:
On a other Windows-PC i have the same issue. qt gui range crash the gnuradio. It seems there is a fault in the windows installation package

ryanvolz, 2023-07-18:
Hi @Tobi82m! I tried this again just now, and I wasn't able to make GRC crash when adding the "QT GUI Range" or any other Qt block in my Windows installation. I'm at a loss for what's happening, and why it would happen for you with the "QT GUI Range" block in particular. Any additional information you could provide might prove helpful.

Tobi82m, 2023-07-27:
On a third Windows 10 PC it works. I have no idea were is the error on the other 2. On the two others there is a lot of programming stuff like arduino ide, visual studio code, mingw, stm32cube ide and so on. Is there a log where i can see what happend?