radioconda/radioconda-installer #49: Missing object attribute in gfsk.py

Gemikro1, 2023-02-06:
Using the GFSK Demod block throws an error in the function `def _print_verbage(self):` because the attribute `_mu` is missing. This attribute seems to have been deprecated but is still used in the function:

```
    def _print_verbage(self):
        print("bits per symbol = %d" % self.bits_per_symbol())
        print("Symbol Sync M&M omega = %f" % self._omega)
        print("Symbol Sync M&M gain mu = %f" % self._gain_mu)
        print("M&M clock recovery mu (Unused) = %f" % self._mu)
        print("Symbol Sync M&M omega rel. limit = %f" %
              self._omega_relative_limit)
        print("frequency error = %f" % self._freq_error)
```

Removing line 273  `print("M&M clock recovery mu (Unused) = %f" % self._mu)` fixes the error ...

ryanvolz, 2023-02-06:
This appears to be a bug with [current GNU Radio](https://github.com/gnuradio/gnuradio/blob/main/gr-digital/python/digital/gfsk.py#L273). I guess not many people have been using that block. I'll suggest that you report this [upstream](https://github.com/gnuradio/gnuradio/issues) to see how it can be handled properly.

Gemikro1, 2023-02-07:
Agreed, i will report it there.
I found that this happens anyway only when verbose output is enabled.

ryanvolz, 2023-04-12:
Fixed by https://github.com/gnuradio/gnuradio/pull/6518, in the `gnuradio` 3.10.6.0 release.

Gemikro1, 2023-04-12:
Perfect, thanks. Will try next time.