radioconda/radioconda-installer #50: Hermes Lite 2 Transceiver Support

GW4ABP-P, 2023-02-07:
Hi Ryan,

Could you possibly add support for the Hermes Lite2. This is a popular amateur radio kit transceiver.

I have the necessary OOT blocks running on GR3.10 under Ubuntu 22.04LTS on a PC and a Pi4.

The block code can be found at the address below: 

https://github.com/daniestevez/gr-hermeslite2

Daniel (the author) is also a contributor to GNU Radio Satellites.

It would be nice to have this running on Windows.

Mike G4ABP

ryanvolz, 2023-02-07:
Hey @daniestevez, is it possible to tag a release of gr-hermeslite2? Anything I should know about packaging this?

daniestevez, 2023-02-07:
Hi Ryan, I've just tagged a release of gr-hermeslite2 (v1.0.0.0). I don't know if this will build under Windows. It uses some UNIX networking stuff, including `ioctl()` (see [this function](https://github.com/daniestevez/gr-hermeslite2/blob/main/lib/metis.cc#L112) for an example).

ryanvolz, 2023-04-07:
I finally took a look at this today. It was pretty easy to get the conda recipe working for Linux and macOS, but yeah [Windows is a problem](https://github.com/ryanvolz/gr-hermeslite2/actions/runs/4640388391/jobs/8212277752) because of the networking. I'm afraid I don't have time to work on porting it myself, so for now I'll just move ahead with making non-Windows packages. Since there is a tagged release (thanks @daniestevez!), I'll also submit the recipe to conda-forge at some point.

I'll be sending a PR with the completed recipe shortly. If anyone wants to hack on a Windows port, enabling CI building/testing on Windows is an easy add-on that could be helpful for getting a port to compile.

kf4mot, 2026-03-11:
It's safe to assume this Windows port is still on pause.
I just wanted to poke to see is the networking problem situation has changed in the couple years since.
Thanks!