radioconda/radioconda-installer #51: latest radioconda - Error

xjugs, 2023-02-09:
Hello,

when I try to install the latest radioconda version (with conda):
conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconda

I got the the following error:
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - radioconday

Is it because there is a "failing" symbol near the "build radioconda" label?

Thanks

ryanvolz, 2023-02-09:
Hmm, it worked for me just now. What platform are you using (e.g. Linux, Windows), and what version of conda? The PackagesNotFoundError referencing "radioconday" seems to indicate a typo somewhere.

The build failing thing was an incidental GitHub issue; it didn't effect the actual release. I restarted an old job to see if that will clear up the badge.

ryanvolz, 2023-02-09:
> BTW, is there any difference between installing radioconda from your channel or via the *.sh file [radioconda-2023.01.19-Linux-x86_64.sh]?

You may get a slightly different overall set of packages installed with the former, since it only has requirements on the [subset of packages of interest](https://github.com/ryanvolz/radioconda/blob/main/installer_specs/radioconda-linux-64.yml) and the dependencies of those packages may have been updated since the time the installer was created. But mostly it should be the same experience. [Installing from the lock file](https://github.com/ryanvolz/radioconda#install-a-particular-release) will produce identical results to the .sh installer and bypasses the conda environment solver.

xjugs, 2023-02-12:
> Anyway, I'll try again (now the radioconda build status is green).

I've checked it again but  the installation still fails. It says:

(base) samla@PC-NUC1:~$ conda create -n radioconda -c conda-forge -c ryanvolz --only-deps radioconday
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - radioconday

Current channels:

  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://conda.anaconda.org/ryanvolz/linux-64
  - https://conda.anaconda.org/ryanvolz/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

Do you have any idea please?

I'm using Ubuntu 20.04.
Anaconda3-2022.10-Linux-x86_64