radioconda/radioconda-installer #55: Python return code 3221225501

LNAppert, 2023-03-09:
Hello,
I am an IT of a school and have an issue with some computers: the installation is done without error or warning, I can run Gnu Radio companion, I open grc file and when I run it some computers return:
>>> Done (return code 3221225501)
without opening the window with the result. I can't explain, I have tried with logs and step by step but no error... Is there a way to show more logs to find what is wrong? I have attached the bad and good result:
[image]
[image]

ryanvolz, 2023-03-09:
This sounds like it could be #25. Does it only stop with that error when a Frequency Sink or Waterfall Sink block is used?

LNAppert, 2023-03-10:
Thank you for your answer, it help me a lot! By replacing fftw3.dll and fftw3f.dll with [this older version](https://anaconda.org/conda-forge/fftw/3.3.9/download/win-64/fftw-3.3.9-nompi_hd3ad3c4_101.tar.bz2) it work now.
This issue happened with Pentium Gold but not on i3 processors.
I'm sorry but I can't answer to your question about Frequency Sink and Waterfall Sink, I am not a end user I just install the app. But I can try if you have exemples?

ryanvolz, 2023-03-14:
If replacing the fftw DLLs fixed it, then it does only apply to blocks that use the fft like the Frequency and Waterfall sinks. I'll close this now since it's a confirmed duplicate.

closed 2023-03-14 by ryanvolz