radioconda/radioconda-installer #56: Pluto not working with Radioconda 3.10.5.0 on Windows 10

GW4ABP-P, 2023-03-10:
Hi Ryan,

I have a Pluto based Flowgraph that does not work on Radioconda Windows 10 using GRC 3.10.5.0, but the same Pluto Flowgraph does work on the same PC if I bring up Ubuntu 22.04 via an Oracle VM and run GRC 3.10.2. (the PC Ethernet connection port remains the same with the same IP address)

This Pluto Flowgraph also works OK with the same Pluto, IP connected to a Raspberry Pi4 running Ubuntu 22.04 with GRC 3.10.2.

The Pluto works with SDR Console under Windows 10 on this PC.
If I then immediately try my Pluto Flowgraph on this PC, with Windows 10 Radioconda it does not work.
If I then immediately try SDR Console again, the Pluto works OK.

Do I need to install the following to make the Pluto work, or should this already be on my system as part of the initial Radioconda install??

https://anaconda.org/conda-forge/gnuradio-iio

I installed the ADI Pluto drivers as mentioned at:

https://github.com/ryanvolz/radioconda#iio-pluto-sdr.

I think they were originally installed 2 years ago, but I re-installed them to make sure. 

The reported errors from the recently installed Radiconda GRC 3.10.5.0 are as follows:

baseband_Pluto_SAT3.py", line 371, in __init__
self.iio_pluto_source_0 = iio.fmcomms2_source_fc32('ip:192.168.0.40' if 'ip:192.168.0.40' else iio.get_pluto_uri(), [True, True], 16384)
AttributeError: module 'gnuradio.iio' has no attribute 'fmcomms2_source_fc32'

>>> Done (return code 1)

I found a file iio.fmcomms2_source.block.yml in the block Library

Would much appreciate your help to sort this one out.

Regards,

Mike, G4ABP

ryanvolz, 2023-03-10:
Hi Mike,

This is a bug with GR 3.10.5.0 that is fixed with GR 3.10.5.1. You can [upgrade](https://github.com/ryanvolz/radioconda#upgrade) to fix it.

Cheers,
Ryan

closed 2023-03-10 by ryanvolz

GW4ABP-P, 2023-03-10:
Ryan, thanks for the very rapid response. Mike