radioconda/radioconda-installer #57: Is there any instruction for creating OOT module in Windows 10?

yourarchipelago, 2023-03-14:
The instruction in GnuRadio website is totally for Linux user which is very different from Windows - Radioconda.

ryanvolz, 2023-03-14:
The procedure is essentially the same, when running from the "Conda Prompt" installed by radioconda. You can use the `gr_modtool` command as the [tutorial](https://wiki.gnuradio.org/index.php?title=Creating_C%2B%2B_OOT_with_gr-modtool) describes. When it comes to compiling the module, you might need to also install `gnuradio-build-deps` (i.e. run `mamba install gnuradio-build-deps`) and pass additional flags to CMake as described [here](https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio).

yourarchipelago, 2023-04-26:
I tried to port a python OOT module in Windows 10, it was easy and sound.
However when it came to c++ OOT module, things went wrong. The project was compiled and built successfully, then I copied .dll and .lib file into "%CONDA_PREFIX%\Library" and the yml file into "%CONDA_PREFIX%\Library\share\gnuradio\grc\blocks\", but when the module was called in Gnuradio, it seemed that Gnuradio was running a python source file, so the c++ OOT would not be called. 

Since there is no instrutions and documents, I really don't know how to port a c++ OOT module.

ryanvolz, 2023-04-26:
When you say that you copied files into the `%CONDA_PREFIX%`, do you mean that you did this manually? Or did you install them by running `cmake --build . --target install`? If actually manually, first I would not recommend that, and second, the `.dll` should go into `%CONDA_PREFIX%\Library\bin` and the `.lib` should go into `%CONDA_PREFIX%\Library\lib`.

But you didn't mention Python bindings for your OOT module, which is usually how one would connect GNU Radio companion to a C++ block (GRC makes a python script that uses the python bindings for the block which use the C++ block library). Did you run `gr_modtool bind` to create Python bindings? (You probably need to `pip install pygccxml` for automatic binding to work, but I don't know for sure as I've never created a C++ OOT module from scratch.) And of course the Python bindings would need to be installed with `cmake --build . --target install`.

You might have better luck asking about the OOT creation process on the [GNU Radio chat](https://wiki.gnuradio.org/index.php/Chat) or [mailing list](https://lists.gnu.org/mailman/listinfo/discuss-gnuradio) since my experience there is lacking. I only really know how to provide the tools to facilitate the process, not the details of the process.

yourarchipelago, 2023-06-01:
Sorry, it has been a long time since the issue was initialized.

yourarchipelago, 2023-06-01:
I tried all these methods above but found that the dll and lib will be targeted at the directory - "c:\programe files\.....". 
I think the position of files is irrelavent, the real thing is that the "dlls" cannot be called in Radioconda.

ryanvolz, 2023-07-24:
Can you walk me through all of the exact steps that you are taking? I think you're probably missing calling `cmake` with arguments like `-DCMAKE_INSTALL_PREFIX="%CONDA_PREFIX%\Library" -DCMAKE_PREFIX_PATH="%CONDA_PREFIX%\Library" -DGR_PYTHON_DIR="%CONDA_PREFIX%\Lib\site-packages"`, but I don't know that for sure without knowing the full procedure of what you're doing.

ryanvolz, 2024-05-09:
I think this is somewhat covered by https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio, so I'm closing to clean up the issue tracker. Please feel free to re-open if you have a follow-up.

closed 2024-05-09 by ryanvolz