radioconda/radioconda-installer #58: windows: gr-fosphor / fosphor_python 'DLL load failed, the specified module could not be found'

carlfarrington, 2023-03-19:
Hi, on Windows 11.

osmocom_fft -F, and also manually trying to use fosphor, results in this (with PYTHONVERBOSE) set:
```
>>> from fosphor_python import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 674, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1176, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: DLL load failed while importing fosphor_python: The specified module could not be found.
```

I am guessing this might be that the module is built against a certain vendor's OpenCL library that I do not have?
I am on an Intel laptop - 8th Gen i5-u (i5-8365U) with UDH Graphics 620. The Intel website says that the OpenCL runtime is built-in to the graphics driver distribution, so nothing to download.

I have previously used gr-fosphor on my i5-5300u Thinkpad, under Linux, with OpenCL-Intel-CPU, and it worked very well. I would like to be able to use it on my Windows laptop though.

Any ideas how I can get it working? Or some guidance on building from source? The notes on the gr-fosphor page suggest it's challenging to build on Windows.

cheers,
Carl

carlfarrington, 2023-03-19:
with dlltracer:

```
In [6]: with dlltracer.Trace(out=sys.stdout):
   ...:         from fosphor.fosphor_python import *
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Lib\site-packages\gnuradio\fosphor\fosphor_python.cp310-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\vcruntime140_1.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\msvcp140.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\gnuradio-fosphor.dll
LoadLibrary \Device\HarddiskVolume3\Users\CarlFarrington\.conda\pkgs\gnuradio-core-3.10.5.1-py310h0fc5001_3\Library\bin\gnuradio-runtime.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\volk.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\gnuradio-pmt.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\OpenCL.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\opengl32.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\freetype.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\boost_thread.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\spdlog.dll
LoadLibrary \Device\HarddiskVolume3\Users\CarlFarrington\.conda\pkgs\gnuradio-core-3.10.5.1-py310h0fc5001_3\Library\bin\gnuradio-fft.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\boost_program_options.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\glew32.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\glfw3.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\fmt.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\mpir.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5OpenGL_conda.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Core_conda.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Gui_conda.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Widgets_conda.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\glu32.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\libpng16.dll
LoadLibrary \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\fftw3f.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\OpenCL.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\libpng16.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\freetype.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\fftw3f.dll
Failed \Device\HarddiskVolume3\Users\CarlFarrington\.conda\pkgs\gnuradio-core-3.10.5.1-py310h0fc5001_3\Library\bin\gnuradio-fft.dll
Failed \Device\HarddiskVolume3\Windows\System32\opengl32.dll
Failed \Device\HarddiskVolume3\Windows\System32\glu32.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\glew32.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\glfw3.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5OpenGL_conda.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Widgets_conda.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Gui_conda.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\Qt5Core_conda.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\gnuradio-fosphor.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\gnuradio-pmt.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\volk.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\boost_thread.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\spdlog.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\boost_program_options.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\fmt.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Library\bin\mpir.dll
Failed \Device\HarddiskVolume3\Users\CarlFarrington\.conda\pkgs\gnuradio-core-3.10.5.1-py310h0fc5001_3\Library\bin\gnuradio-runtime.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\msvcp140.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\vcruntime140_1.dll
Failed \Device\HarddiskVolume3\ProgramData\radioconda\Lib\site-packages\gnuradio\fosphor\fosphor_python.cp310-win_amd64.pyd
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
Cell In[6], line 2
      1 with dlltracer.Trace(out=sys.stdout):
----> 2         from fosphor.fosphor_python import *
​
File C:\ProgramData\radioconda\Lib\site-packages\gnuradio\fosphor\__init__.py:9
      1 # Copyright 2008-2020 Free Software Foundation, Inc.
      2 # Copyright 2013-2021 Sylvain Munaut <tnt@246tNt.com>
      3 #
   (...)
      7
      8 # import pybind11 generated symbols into the iqbalance namespace
----> 9 from .fosphor_python import *
​
ImportError: DLL load failed while importing fosphor_python: The specified module could not be found.
```

carlfarrington, 2023-03-20:
hmm. After lots of faffing around trying (and failign) to compile gr-fsphor with Intel C compiler and VS build tools, and then vc, I simply removed gnuradio-fosphor from the radioconda conda environment, and then installed a slightly older version, and it works!

I installed gnuradio-fosphor=0.0.0.20221223.dev+g84e9d8f

closed 2023-03-20 by carlfarrington

ryanvolz, 2023-03-21:
I compared the two `gnuradio-fosphor` packages and couldn't identify a major difference that could cause this, but I think there are other possible explanations. Can you tell what OpenCL implementation is being used when you run gr-fosphor now? In attempting to compile gr-fosphor yourself, did you happen to install any OpenCL vendor libraries on your system? And perhaps the simplest explanation: there are subtle differences between installing a package through an installer like radioconda and installing using the conda/mamba command, so maybe using the latter approach is what actually got it working? Have you tried installing the latest version now with conda/mamba?