radioconda/radioconda-installer #59: Error install spyder

xjugs, 2023-03-27:
when I try install spyder from anaconda-navigator within radioconda environment, I got an error:
"Spyder cannot be installed on this environment".

However, in other enviroments, I do success to install spyder. the problem exist just within radioconda environment.

I am using:
ubuntu 20.04
radioconda 2023.02.24 (the latest one)

ryanvolz, 2023-03-27:
What happens if you try to install it from the command line, e.g. `mamba install spyder` from a terminal with the radioconda environment activated? If there's a package conflict, that should tell us.

ryanvolz, 2023-03-28:
I don't know why there's a conflict, and what it claims is a conflict doesn't make sense to me (spyder needs ipython which needs __linux, but it says you have __linux in the line above that?). I'm able to install `spyder` in my radioconda environment without issues. Can you try installing it with `mamba` instead of `conda`? The different solver might give a better message. You'd need to have mamba installed in your base environment to use it.

xjugs, 2023-03-29:
It indeed worked with mamba. But why? How is it differ from conda?

xjugs, 2023-03-31:
Ok. Thank you

closed 2023-07-18 by ryanvolz