radioconda/radioconda-installer #61: installing additional package: gr-plasma

eduemon, 2023-04-04:
Dear radioconda,
I am new with radioconda and I have just installed gnu radio using radioconda installer. I want to install new package. Can you please add this package into your installer or is there any tutorial to install new package into current gnu-radio?
Thank you

gr-plasma link: https://github.com/ShaneFlandermeyer/gr-plasma

ryanvolz, 2023-04-04:
Generally you can build any OOT modules that I haven't managed to package from source: https://wiki.gnuradio.org/index.php/CondaInstall#Building_OOT_modules_to_use_with_conda-installed_GNU_Radio.

But I see that gr-plasma depends on plasma-dsp which depends on arrayfire, all of which are not available as conda packages, so that's a lot to build. It looks like a cool package though, so I'll keep it in mind.

eduemon, 2023-04-04:
Thank you. Can't wait to see it included in the next version. thank you for your reply.