radioconda/radioconda-installer #63: Problem with fosphor sink in GNUradio companion for Windows

alexlitv277, 2023-04-11:
Hi ALL!
I have problem with fosphor sink in GNUradio companion in latest version for Windows 3.10.6.0 :

File "F:\grc\companion382\top_block.py", line 14, in <module>
    from gnuradio import fosphor
  File "C:\Users\litv\AppData\Local\radioconda\lib\site-packages\gnuradio\fosphor\__init__.py", line 9, in <module>
    from .fosphor_python import *
ImportError: DLL load failed while importing fosphor_python:

Last working version with fosphor sink for windows in my case was 3.8.2 with nonconda installer.

See my details : 
conda info --all

     active environment : base
    active env location : C:\Users\litv\AppData\Local\radioconda
            shell level : 1
       user config file : C:\Users\litv\.condarc
 populated config files : C:\Users\litv\AppData\Local\radioconda\.condarc
                          C:\Users\litv\.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=11.4=0
                          __win=0=0
       base environment : C:\Users\litv\AppData\Local\radioconda  (writable)
      conda av data dir : C:\Users\litv\AppData\Local\radioconda\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/ryanvolz/win-64
                          https://conda.anaconda.org/ryanvolz/noarch
          package cache : C:\Users\litv\AppData\Local\radioconda\pkgs
                          C:\Users\litv\.conda\pkgs
                          C:\Users\litv\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\litv\AppData\Local\radioconda\envs
                          C:\Users\litv\.conda\envs
                          C:\Users\litv\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.3.1 requests/2.28.2 CPython/3.10.10 Windows/10 Windows/10.0.19044
          administrator : True
             netrc file : None
           offline mode : False

# conda environments:
#
base                  *  C:\Users\litv\AppData\Local\radioconda

sys.version: 3.10.10 | packaged by conda-forge | (mai..
sys.prefix: C:\Users\litv\AppData\Local\radioconda
sys.executable: C:\Users\litv\AppData\Local\radioconda\python.exe
conda location: C:\Users\litv\AppData\Local\radioconda\lib\site-packages\conda
conda-build: None
conda-env: C:\Users\litv\AppData\Local\radioconda\Scripts\conda-env.exe
user site dirs:

CIO_TEST: <not set>
CONDA_DEFAULT_ENV: base
CONDA_EXE: C:\Users\litv\AppData\Local\radioconda\condabin\..\Scripts\conda.exe
CONDA_EXES: "C:\Users\litv\AppData\Local\radioconda\condabin\..\Scripts\conda.exe"
CONDA_PREFIX: C:\Users\litv\AppData\Local\radioconda
CONDA_PROMPT_MODIFIER: (base)
CONDA_PYTHON_EXE: C:\Users\litv\AppData\Local\radioconda\python.exe
CONDA_ROOT: C:\Users\litv\AppData\Local\radioconda
CONDA_SHLVL: 1
CURL_CA_BUNDLE: <not set>
HOMEPATH: \Users\litv
HTTPS_PROXY: <set>
HTTP_PROXY: <set>
LD_PRELOAD: <not set>
PATH: C:\Users\litv\AppData\Local\radioconda;C:\Users\litv\AppData\Local\radioconda\Library\mingw-w64\bin;C:\Users\litv\AppData\Local\radioconda\Library\usr\bin;C:\Users\litv\AppData\Local\radioconda\Library\bin;C:\Users\litv\AppData\Local\radioconda\Scripts;C:\Users\litv\AppData\Local\radioconda\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0;C:\Windows\System32\OpenSSH;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\MATLAB\R2022b\runtime\win64;C:\MATLAB\R2022b\bin;C:\MATLAB\R2022a\runtime\win64;C:\MATLAB\R2022a\bin;d:\MATLAB\R2013b\runtime\win64;d:\MATLAB\R2013b\bin;d:\Program Files\Microsoft VS Code\bin;d:\Program Files\UHD\bin;C:\Program Files\bladeRF\x64;C:\Users\litv\radioconda\condabin;C:\Users\litv\AppData\Local\Microsoft\WindowsApps;.
PSMODULEPATH: C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
PTPATH: C:\Users\litv\AppData\Local\ProjectTree
REQUESTS_CA_BUNDLE: <not set>
SSL_CERT_FILE: <not set>
VBOX_MSI_INSTALL_PATH: C:\Program Files\Oracle\VirtualBox\

I will glad to any ideas to use fosphor with conda installation.

Best Regards
Alexander

ryanvolz, 2023-04-11:
Hmm, I just tried this latest package in my Windows VM and it worked fine for me, so nothing has changed there. 

> File "F:\grc\companion382\top_block.py", line 14, in
> from gnuradio import fosphor
> File "C:\Users\litv\AppData\Local\radioconda\lib\site-packages\gnuradio\fosphor__init__.py", line 9, in
> from .fosphor_python import *
> ImportError: DLL load failed while importing fosphor_python:

We need to figure out what DLL is failing to load and why. Can you try navigating to `C:\Users\litv\AppData\Local\radioconda\lib\site-packages\gnuradio\fosphor` and running `python -c "import fosphor_python"`?

alexlitv277, 2023-04-12:
Hi Ryan! Thank you for your job. 
I repeated the installation on three Windows computers and got the same bad results with fosphor sink as described already in the question.

For example, for your check:

C:\radioconda\Lib\site-packages\gnuradio\fosphor>python -c "import fosphor_python"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: DLL load failed while importing fosphor_python:

But on the fourth computer it suddenly worked (the only difference is - it has a video card not Nvidia but Intel) and I can see 
lovely real-time spectrum.
I have no ideas , from one distributive - different situations.

Best Regards
Alexander

ryanvolz, 2023-04-12:
Nvidia vs Intel being a difference is interesting and definitely something to dig into. We're using `khronos-opencl-icd-loader` on Windows, and [this issue](https://github.com/conda-forge/khronos-opencl-icd-loader-feedstock/issues/11) and [this discussion](https://github.com/inducer/pyopencl/discussions/524) might describe what's happening. It's possible that the Nvidia ICD loader has higher priority on your path, so it gets used instead, and then it fails because `gnuradio-fosphor` was not compiled against that. You can test this by seeing if you have an `OpenCL.dll` somewhere in `C:\System32` and then temporarily renaming it.

ryanvolz, 2023-04-12:
Also I want to note the similarity to #58 just so GitHub will show links between the two issues.

shinchik17, 2023-08-29:
Hi, guys! I finally found a solution of this problem during long digging into. The problem is absence of two dlls:
ucrtbased.dll and vcruntime140d in System32 folder. These components are part of Microsoft Visual Studio, that's why developers such as @ryanvolz haven't this trouble at all. After adding this libraries to System32 fosphor sink works fine. I use GR 3.10.7 and Windows 10

Biran1911, 2023-08-31:
> Hi, guys! I finally found the solution of this problem during long digging into. The problem is absence of two dlls: ucrtbased.dll and vcruntime140d.dll in System32 folder. These components are part of Microsoft Visual Studio, that's why developers such as @ryanvolz haven't this trouble at all. After adding this libraries to System32 a fosphor sink works fine. I use GR 3.10.7 and Windows 10

I am having the exact issue.
would you kindly describe to process? I tried to install Microsoft visual studio 2013, 2015-2022.
in my system32 i see the following files:
ucrtbase.dll and vcruntime140.dll

shinchik17, 2023-08-31:
> I am having the exact issue. would you kindly describe to process? I tried to install Microsoft visual studio 2013, 2015-2022. in my system32 i see the following files: ucrtbase.dll and vcruntime140.dll

Check names of dlls I wrote and yours. They have letter "D" at the end. It's a differrent dlls. Firstly check if you have these dlls with "D" in your System32 folder.

Biran1911, 2023-08-31:
> > I am having the exact issue. would you kindly describe to process? I tried to install Microsoft visual studio 2013, 2015-2022. in my system32 i see the following files: ucrtbase.dll and vcruntime140.dll
> 
> Check names of dlls I wrote and yours. Mine have letter "D" at the end. It's a differrent dlls. Firstly check if you have these dlls with "D" in your System32 folder.

I understand what you wrote, i dont got the debug version of those dll's.
Im simply asking you how did you obtain yours? for example is there any file to download?
Thanks :)

shinchik17, 2023-09-01:
> I understand what you wrote, i dont got the debug version of those dll's. Im simply asking you how did you obtain yours? for example is there any file to download? Thanks :)

I got you :) Try these ones
[gr310dll.zip](https://github.com/ryanvolz/radioconda/files/12495290/gr310dll.zip)

Biran1911, 2023-09-01:
> > I understand what you wrote, i dont got the debug version of those dll's. Im simply asking you how did you obtain yours? for example is there any file to download? Thanks :)
> 
> I got you :) Try these ones [gr310dll.zip](https://github.com/ryanvolz/radioconda/files/12495290/gr310dll.zip)

Thank you so much :)

ryanvolz, 2023-09-05:
Nice digging @shinchik17! Question to both you and @Biran1911: are you using an Nvidia graphics card like @alexlitv277, or something else like Intel or AMD?

If either of you is seeing this without an Nvidia card and its OpenCL ICD loader software, then that points to a different issue than those linked in https://github.com/ryanvolz/radioconda/issues/63#issuecomment-1505404450 and potentially something I can get fixed while still waiting on resolution there.

If you're both using Nvidia, that's still very useful to know. The workaround is very helpful regardless. Thanks!

shinchik17, 2023-09-05:
> Nice digging @shinchik17! Question to both you and @Biran1911: are you using an Nvidia graphics card like @alexlitv277, or something else like Intel or AMD?
> 
> If either of you is seeing this without an Nvidia card and its OpenCL ICD loader software, then that points to a different issue than those linked in [#63 (comment)](https://github.com/ryanvolz/radioconda/issues/63#issuecomment-1505404450) and potentially something I can get fixed while still waiting on resolution there.
> 
> If you're both using Nvidia, that's still very useful to know. The workaround is very helpful regardless. Thanks!

I've also tried on the second PC with AMD Radeon integrated graphics. The issue with fosphor is the same. The solution is also the same - adding 2 dlls works.

Biran1911, 2023-09-06:
>

> Nice digging @shinchik17! Question to both you and @Biran1911: are you using an Nvidia graphics card like @alexlitv277, or something else like Intel or AMD?
>
> If either of you is seeing this without [...]

Hey @ryanvolz, I tried it on a desktop computer with AMD RX 6800 XT GPU and on a laptop with Intel Iris Xe GPU.
In both cases the debug dll's did the trick, thanks to @shinchik17 again for the huge help.

ryanvolz, 2023-11-06:
Thanks again for the investigation and testing @shinchik17 and @Biran1911! I tracked down the need for the debug DLLs (ucrtbased.dll and vcruntime140d.dll) to the `glfw` package, which had mistakenly been compiled in "Debug" mode instead of "Release" mode. I alerted the maintainers and got a fix merged, so the latest version should no longer need those DLLs. I'll be making a new radioconda release shortly (waiting on one more thing) with the fixed package, and that should resolve this issue. If you could upgrade and test manually before the release, that would be wonderful, but I'll leave this open until someone can confirm that the problem is resolved.

ryanvolz, 2024-05-09:
I'll assume this is fixed now.

closed 2024-05-09 by ryanvolz