radioconda/radioconda-installer #64: Pausing the flowgraph freezes the app on Windows

TheCrazyT, 2023-05-01:
I'm using a windows environment.

I just noticed that you can't use a variable in the "Run" property under "Options" of your flowgraph.
[image]

I normaly use a checkbox-variable to pause the flowgraph.
This is useful for timesinks etc. (if you want to measure time differences).
In GnuRadio version 3.9 this worked without a problem.
But on latest 3.10 the whole window just freezes, without beeing able to do anything (looks like it does not react on any window event at all).

TheCrazyT, 2023-05-01:
I'm just using a toggle button now ... for some weird reason that does not seem to have that problem ...

ryanvolz, 2023-05-04:
This feels like it could be an upstream [gnuradio](https://github.com/gnuradio/gnuradio/issues) bug, but I do know that the use case you're describing works for me with GR 3.10 on Linux. If I get some time I can try it on a Windows VM. But overall you might get more traction by filing this bug report directly with GNU Radio.

ryanvolz, 2023-07-18:
I tried this out finally, and was able to pause and restart a simple flowgraph on Windows using the checkbox widget. Maybe some upgrade to Qt fixed this in the meantime? Or maybe there is a particular block in your flowgraph that doesn't like being paused with GR 3.10?

ryanvolz, 2023-07-24:
So the "aO" is audio overrun, which sounds to me like the audio device is still trying to send data to the GR audio source but can't because the flowgraph is paused. One thing that I know has changed with GR 3.9.1.0 is that the preferred audio source on Windows is now Portaudio instead of the (barely maintained) Windows native audio. It looks like GR's Portaudio backend doesn't handle pausing correctly, but the Windows native audio backend does. You can test this by editing `C:\Users\<YOUR_USERNAME>\AppData\Roaming\.gnuradio\config.conf` (or creating it if it doesn't exist) and adding the following:
```
[audio]
audio_module = windows
```
For me, that gets rid of the "aO", and maybe it fixes your freezing too.

(If you don't have a `C:\Users\<YOUR_USERNAME>\AppData\Roaming\.gnuradio` directory, it's possible that it should be something different on your system. Check what that would be by running `gnuradio-config-info --userprefsdir`.)

The fact that the Portaudio backend produces "aO" when paused might constitute an upstream GNU Radio bug, assuming that's not what was intended.

closed 2023-07-25 by TheCrazyT