radioconda/radioconda-installer #71: cmake -G Xcode ... fails

kgarrels, 2023-09-03:
Hi,

I recently switched from macports to radioconda for gnuradio.

I have a cmake based project that builds fine with It does work correctly if I just build with 
`cmake -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_PREFIX_PATH=$CONDA_PREFIX -DLIB_SUFFIX="" ..`

But now, when I try to generate an Xcode project from a cmake project with 
`cmake -G Xcode -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_PREFIX_PATH=$CONDA_PREFIX -DLIB_SUFFIX="" ..`
it fails, with errors.

I want the xcode version for debugging.

Errors are:
```cmake -G Xcode -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX -DCMAKE_PREFIX_PATH=$CONDA_PREFIX -DLIB_SUFFIX="" ..
-- The C compiler identification is unknown
-- The CXX compiler identification is Clang 15.0.7
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/kai/radioconda/bin/x86_64-apple-darwin13.4.0-ld
-- Check for working C compiler: /Users/kai/radioconda/bin/x86_64-apple-darwin13.4.0-ld - broken
CMake Error at /Users/kai/radioconda/share/cmake-3.26/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/Users/kai/radioconda/bin/x86_64-apple-darwin13.4.0-ld"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5

    Run Build Command(s):/usr/bin/xcodebuild -project CMAKE_TRY_COMPILE.xcodeproj build -target cmTC_88fa8 -parallelizeTargets -configuration Debug -hideShellScriptEnvironment && Command line invocation:
        /Applications/Xcode.app/Contents/Developer/usr/bin/xcodebuild -project CMAKE_TRY_COMPILE.xcodeproj build -target cmTC_88fa8 -parallelizeTargets -configuration Debug -hideShellScriptEnvironment

    User defaults from command line:
        HideShellScriptEnvironment = YES
        IDEPackageSupportUseBuiltinSCM = YES

    Computing target dependency graph and provisioning inputs

    Create build description
    Build description signature: 6522802e27f4d99a579aeff48583c727
    Build description path: /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/XCBuildData/6522802e27f4d99a579aeff48583c727-desc.xcbuild

    note: Building targets in dependency order
    CreateBuildDirectory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/CMAKE_TRY_COMPILE.xcodeproj
        builtin-create-build-directory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build

    CreateBuildDirectory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/CMAKE_TRY_COMPILE.xcodeproj
        builtin-create-build-directory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug

    CreateBuildDirectory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/EagerLinkingTBDs
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/CMAKE_TRY_COMPILE.xcodeproj
        builtin-create-build-directory /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/EagerLinkingTBDs

    WriteAuxiliaryFile /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/cmTC_88fa8.LinkFileList (in target 'cmTC_88fa8' from project 'CMAKE_TRY_COMPILE')
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5
        write-file /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/cmTC_88fa8.LinkFileList

    CompileC /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/testCCompiler.o /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/testCCompiler.c normal x86_64 c com.apple.compilers.llvm.clang.1_0.compiler (in target 'cmTC_88fa8' from project 'CMAKE_TRY_COMPILE')
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5
        x86_64-apple-darwin13.4.0-clang -x c -target x86_64-apple-macos13.1 -fmessage-length\=0 -fdiagnostics-show-note-include-stack -fmacro-backtrace-limit\=0 -Wno-trigraphs -fpascal-strings -O2 -Wno-missing-field-initializers -Wno-missing-prototypes -Wno-return-type -Wno-missing-braces -Wparentheses -Wswitch -Wno-unused-function -Wno-unused-label -Wno-unused-parameter -Wno-unused-variable -Wunused-value -Wno-empty-body -Wno-uninitialized -Wno-unknown-pragmas -Wno-shadow -Wno-four-char-constants -Wno-conversion -Wno-constant-conversion -Wno-int-conversion -Wno-bool-conversion -Wno-enum-conversion -Wno-float-conversion -Wno-non-literal-null-conversion -Wno-objc-literal-conversion -Wshorten-64-to-32 -Wpointer-sign -Wno-newline-eof -Wno-implicit-fallthrough -DCMAKE_INTDIR\=\"Debug\" -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -fasm-blocks -fstrict-aliasing -Wdeprecated-declarations -Wno-sign-conversion -Wno-infinite-recursion -Wno-comma -Wno-block-capture-autoreleasing -Wno-strict-prototypes -Wno-semicolon-before-method-body -I/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug/include -I/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/DerivedSources-normal/x86_64 -I/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/DerivedSources/x86_64 -I/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/DerivedSources -F/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug -march\=core2 -mtune\=haswell -mssse3 -ftree-vectorize -fPIC -fPIE -fstack-protector-strong -pipe -isystem /Users/kai/radioconda/include -MMD -MT dependencies -MF /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/testCCompiler.d --serialize-diagnostics /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/testCCompiler.dia -c /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/testCCompiler.c -o /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/testCCompiler.o

    Ld /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug/cmTC_88fa8 normal (in target 'cmTC_88fa8' from project 'CMAKE_TRY_COMPILE')
        cd /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5
        /Users/kai/radioconda/bin/x86_64-apple-darwin13.4.0-ld -target x86_64-apple-macos13.1 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -L/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/EagerLinkingTBDs -L/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug -F/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/EagerLinkingTBDs -F/Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug -filelist /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/cmTC_88fa8.LinkFileList -Wl,-pie -Wl,-headerpad_max_install_names -Wl,-dead_strip_dylibs -Wl,-rpath,/Users/kai/radioconda/lib -L/Users/kai/radioconda/lib -Wl,-search_paths_first -Wl,-headerpad_max_install_names -Xlinker -dependency_info -Xlinker /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/build/cmTC_88fa8.build/Debug/Objects-normal/x86_64/cmTC_88fa8_dependency_info.dat -o /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug/cmTC_88fa8
    ld: unknown option: -target

    ** BUILD FAILED **

    The following build commands failed:
    	Ld /Users/kai/gqrx/xcode/CMakeFiles/CMakeScratch/TryCompile-Hu82I5/Debug/cmTC_88fa8 normal (in target \'cmTC_88fa8\' from project \'CMAKE_TRY_COMPILE\')
    (1 failure)

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
```

I would be very grateful for a hint.

regards,
kai

ryanvolz, 2023-09-05:
What exact packages do you have installed? Try `mamba list`.

CMake is not successfully identifying the correct C compiler (thinking that it's `x86_64-apple-darwin13.4.0-ld`), so it will be useful to know what (if any) extra packages you have installed beyond radioconda. You probably need what conda-forge calls a "compiler activation" package that runs a script to setup the build system when you activate the conda environment. You probably want the `clang_osx-64` package, but if you already have that then the activation script is not doing what you need it to.

kgarrels, 2023-09-05:
thanks.

I have `clang_osx-64` installed - and I think the main riddle is: why does `cmake ..` work nicely, but `cmake -G Xcode ..` fail?
And why is it complaining about `ld`, and not the compiler?

I tried installing/ removing the `compilers`package, no change.

I attach the `mamba list`output, maybe you see something very obvious.

```
mamba list
# packages in environment at /Users/kai/radioconda:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
airspy                    1.0.10               h0d85af4_0    conda-forge
airspyhf                  1.6.8                h0d85af4_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
appnope                   0.1.3              pyhd8ed1ab_0    conda-forge
asciimatics               1.14.0             pyhd8ed1ab_2    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.5              pyhd8ed1ab_0    conda-forge
bcrypt                    4.0.1           py310h3461e44_0    conda-forge
bidict                    0.22.1             pyhd8ed1ab_0    conda-forge
bladerf                   2023.02              hb7f2c08_0    conda-forge
blinker                   1.6.2              pyhd8ed1ab_0    conda-forge
boltons                   23.0.0             pyhd8ed1ab_0    conda-forge
boost-cpp                 1.78.0               hf5ba120_3    conda-forge
brotli                    1.1.0                h0dc2134_0    conda-forge
brotli-bin                1.1.0                h0dc2134_0    conda-forge
brotli-python             1.1.0           py310h9e9d8ca_0    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.19.1               h0dc2134_0    conda-forge
c-compiler                1.6.0                h63c33a9_0    conda-forge
ca-certificates           2023.7.22            h8857fd0_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            h09dd18c_1016    conda-forge
cctools                   973.0.1             hd9ad811_14    conda-forge
cctools_osx-64            973.0.1             habff3f6_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310ha78151a_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
clang                     15.0.7               h694c41f_3    conda-forge
clang-15                  15.0.7          default_hdb78580_3    conda-forge
clang_osx-64              15.0.7               h03d6864_3    conda-forge
clangxx                   15.0.7          default_hdb78580_3    conda-forge
clangxx_osx-64            15.0.7               h2133e9c_3    conda-forge
click                     8.1.7           unix_pyh707e725_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
cmake                     3.26.4               hf40c264_0    conda-forge
codec2                    1.0.3                h0d85af4_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compiler-rt               15.0.7               he1888fc_1    conda-forge
compiler-rt_osx-64        15.0.7               he1888fc_1    conda-forge
conda                     23.7.3          py310h2ec42d9_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
construct                 2.10.68            pyhd8ed1ab_0    conda-forge
contourpy                 1.1.0           py310h88cfcbd_0    conda-forge
cppzmq                    4.10.0               h478c2f3_0    conda-forge
cryptography              41.0.3          py310ha1817de_0    conda-forge
cxx-compiler              1.6.0                h1c7c39f_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
digital_rf                2.6.8           py310h4951aab_2    conda-forge
[... 291 lines cut ...]
typing_extensions         4.7.1              pyha770c72_0    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
uhd                       4.4.0.0         py310h50d9707_0    conda-forge
unicodedata2              15.0.0          py310h90acd4f_0    conda-forge
urllib3                   2.0.4              pyhd8ed1ab_0    conda-forge
volk                      3.0.0                h75ea093_1    conda-forge
watchdog                  3.0.0           py310h389cd99_0    conda-forge
wcwidth                   0.2.6              pyhd8ed1ab_0    conda-forge
websocket-client          1.6.2              pyhd8ed1ab_0    conda-forge
werkzeug                  2.3.7              pyhd8ed1ab_0    conda-forge
wheel                     0.41.2             pyhd8ed1ab_0    conda-forge
wxwidgets                 3.2.2.1              h254b04c_3    conda-forge
[... cut ...]
zeromq                    4.3.4                he49afe7_1    conda-forge
zipp                      3.16.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h8a1eda9_5    conda-forge
zope.event                5.0                pyhd8ed1ab_0    conda-forge
zope.interface            6.0             py310h90acd4f_0    conda-forge
zstandard                 0.19.0          py310h151724a_2    conda-forge
zstd                      1.5.2                h829000d_7    conda-forge
```

kgarrels, 2023-09-05:
I just found: https://stackoverflow.com/questions/65146106/xcodebuild-using-ld-rather-than-clang-for-linking

after `export LD="/usr/bin/clang"`, I was able to create the xcode project.

ryanvolz, 2023-09-05:
Now my theory is that the conda activation script for clang is setting variables that make it not work with the Xcode generator (like [here](https://github.com/conda-forge/clang-compiler-activation-feedstock/blob/main/recipe/activate-clang.sh#L133C5-L133C5) where it sets the linker to ld, when apparently you need clang). Removing the `clang_osx-64` and `clangxx_osx-64` (and maybe any other clang packages) might help by just letting Xcode do its own thing.

I also wonder if there is some path using the options described on the [Xcode generator documentation page](https://cmake.org/cmake/help/latest/generator/Xcode.html) to set the toolset or build system to something compatible with what the conda-forge packages are doing. But the workaround you found is probably easier.

Feel free to reopen this if you run into more problems.

closed 2023-09-05 by ryanvolz