radioconda/radioconda-installer #73: Are you able to update Inspectrum... SigMF

loidmen, 2023-10-04:
SigMF is a new data format for saving IQ recordings. Inspectrum can open those files but only if this addon is activated when compiling.

Is this feature available for windows?

Thanks.

ryanvolz, 2023-10-04:
I think that the lack of SigMF support in the current package is because it is based on the last tagged version, and Inspectrum has had the SigMF updates since then. Conda-forge policy is to only package from tagged versions, so the best way to get the SigMF features into the package would be to kindly ping the upstream developers and ask them to tag a new release. If they don't want to or don't respond after a while, then I can consider doing some manual patching.

loidmen, 2023-10-04:
OK. Thank you.

loidmen, 2023-10-29:
The Inspectrum team has updated the tagged version to include support for SigMF files and annotations. It's version 0.3.1.

Thanks.

ryanvolz, 2023-10-30:
Great! After a Windows build issue, I just merged the package for inspectrum 0.3.1. It will be in the next radioconda release, which should be soon.

closed 2023-10-30 by ryanvolz

loidmen, 2023-10-30:
Yes, I updated inspectrum and everything else too. Thank you so much for your speedy response and for putting so much effort and time into radioconda.
Inspectrum works well now, just by opening the SigMF meta file; it displays the spectrum and all the annotations.

Thank you.