radioconda/radioconda-installer #75: ImportError: cannot import name 'qtgui' from partially initialized module 'gnuradio' (most likely due to a circular import) (C:\Users\rhofmann\Documents\gnuradio.py)

raymundhofmann, 2023-10-12:
```
Generating: 'C:\\Users\\rhofmann\\Documents\\gnuradio.py'

Executing: C:\Users\rhofmann\radioconda\python.exe -u C:\Users\rhofmann\Documents\gnuradio.py

Traceback (most recent call last):
  File "C:\Users\rhofmann\Documents\gnuradio.py", line 13, in <module>
    from gnuradio import qtgui
  File "C:\Users\rhofmann\Documents\gnuradio.py", line 13, in <module>
    from gnuradio import qtgui
ImportError: cannot import name 'qtgui' from partially initialized module 'gnuradio' (most likely due to a circular import) (C:\Users\rhofmann\Documents\gnuradio.py)

>>> Done (return code 1)

```
What is wrong?

ryanvolz, 2023-10-30:
You might be running into an issue from the fact that you've called your flowgraph `gnuradio`, so that when it tries to import the module `gnuradio` it actually finds your `gnuradio.py` first and then fails because it's not the module. Try renaming it to something else?

raymundhofmann, 2023-10-31:
Already renamed the flowgraph to "test" and the ID option to "test", no change. 
No one using it on Windows or something about my machine? 
Anyway, managed to compile gnu radio and run it on WSL2 / Ubuntu22.04 plus Vcxsrv, Pulseaudio, Hackrf tools/access and this is working so far.

ryanvolz, 2023-10-31:
> Already renamed the flowgraph to "test" and the ID option to "test", no change.
Did you happen to delete the `gnuradio.py` file that would still probably be left behind there even after the flowgraph name change? That's probably necessary too.

But if you have a working setup now then I can understand if you don't have the motivation to continue running this down. There are plenty of regular Windows users, probably even the majority, so if it's not the flowgraph name then there's probably something uncommon about your machine that would be good to understand.

raymundhofmann, 2023-10-31:
I deleted the gnuradio.py and it is working now. Quite a trap for the unsuspecting, should be fixed. Anyway, the grc itself is maybe scaled to 70% of the font size of the rest making it a bit inconvenient to use on high DPI screens, while the generated python/QT scales properly like the rest on the system.

ryanvolz, 2023-11-07:
> I deleted the gnuradio.py and it is working now. Quite a trap for the unsuspecting, should be fixed.

I just filed a bug upstream at https://github.com/gnuradio/gnuradio/issues/6945.

> Anyway, the grc itself is maybe scaled to 70% of the font size of the rest making it a bit inconvenient to use on high DPI screens

It's possible that this is related specifically to GTK and how it is built for conda-forge/radioconda, but it's more likely an upstream GRC bug which would not be surprising given other reports with high DPI screens: https://github.com/gnuradio/gnuradio/issues/5841. I don't think any of the developers are using high DPI systems to test on (and I don't have one either!). It's possible that this gets fixed when GRC switches to Qt as planned in the next year. If it's a big annoyance, I'd suggest filing a bug report at https://github.com/gnuradio/gnuradio and include as many details as you can.

closed 2024-05-09 by ryanvolz