radioconda/radioconda-installer #78: ImportError: DLL load failed while importing qtgui_python

ShoeBro, 2023-11-25:
Hello!

I have installed radioconda with the windows installer (3.10.8.0 (Python 3.11.6)), but when running the first tutorial in GNU radio companion I get the error below:

```
packages\gnuradio\qtgui\__init__.py", line 23, in <module>
    from .qtgui_python import *
ImportError: DLL load failed while importing qtgui_python: The specified procedure could not be found.
```

It seems to be a problem for all qt gui cards. I have tried the following:
- Use time sink only
- Use freq sink only
- Uninstalled all other python installs
- Removed Pycharm and Spyder
- Tried previous radioconda version
- Nothing related to python is in the environment variables
- Run the generated .py file in vscode with the radioconda interpreter.

All this has given me the same error and I am out of ideas. Any help is appreciated, thanks!

ryanvolz, 2023-11-27:
Thanks for the report. Evidently some library is failing to load, and we need to figure out what library that is and why. Can you do the following?:

1. Open the "Conda Prompt" as an Administrator by right clicking on the shortcut in the launcher and choosing "Run as administrator".
2. Install dlltracer using the following command: `mamba install -c "conda-forge/label/dlltracer_dev" dlltracer`
3. Open an ipython prompt by running `ipython` from the Conda Prompt
4. Execute the following commands:
```
import dlltracer
import sys

with dlltracer.Trace(out=sys.stdout):
    import gnuradio.qtgui
```

That should print out a list of the DLLs that it imports and what it fails on.

ShoeBro, 2023-11-28:
It is quite strange! So when I run the code you just posted I get the same error, but I found out that it is very dependant on the order of the imports.

So when I run the original code generated by GRC I get the error, with the code written like this:
```
from PyQt5 import Qt 
from gnuradio import qtgui
... 
```

But when I swop the order of the imports, the error does not show, with the code like this:
```
from gnuradio import qtgui
from PyQt5 import Qt
...
```
So for some reason that solves my issue, but why it is an issue in the first place remains a mystery.

ryanvolz, 2023-11-28:
Good to know that my hunch was correct! When you run the import trace on each of those lines individually, I think the output should provide the DLLs that get loaded in the order that they get loaded. If you do that for `from PyQt5 import Qt`, then separately for `from gnuradio import qtgui`, post the full outputs here and we can compare the DLL lists and look for any differences. There's probably two incompatible versions of the same library somewhere, and each of those imports naturally wants to load a different one but will try to use the already-loaded DLL from the first statement when they are done together.

ShoeBro, 2023-11-29:
Great stuff! I did the trace individually, and then threw them into excel to see if there are differences in the order they are loaded as well as if they are loaded compared to one another. Unfortunately nothing stood out to me, and I also don't know enough to know what I am looking for. Here is the print out:

Here is the first one (```from PyQt5 import Qt```)
```
LoadLibrary \Device\HarddiskVolume3\Windows\System32\kernel.appcore.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\Qt.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtXmlPatterns.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Core_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5XmlPatterns_conda.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\ole32.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\combase.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\userenv.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\mpr.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Network_conda.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\shell32.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\icuuc73.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\icuin73.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\pcre2-16.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\vcruntime140_1.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\crypt32.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\netapi32.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\zstd.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\msvcp140.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\winmm.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\msvcp140_1.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\icudt73.dll
Failed \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\icudt73.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\libssl-3-x64.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\libcrypto-3-x64.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\netutils.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\srvcli.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\cryptbase.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\icudt73.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\sip.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtCore.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtNetwork.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtXml.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Xml_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtWinExtras.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5WinExtras_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Widgets_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Gui_conda.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\dwmapi.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\d3d11.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\uxtheme.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\libpng16.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\dxgi.dll
LoadLibrary \Device\HarddiskVolume3\Windows\System32\dxgi.dll
Failed \Device\HarddiskVolume3\Windows\System32\dxgi.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtGui.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtWidgets.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtWebSockets.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5WebSockets_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtWebChannel.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5WebChannel_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Qml_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtTextToSpeech.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5TextToSpeech_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtTest.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Test_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtSvg.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Svg_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtSql.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\Qt5Sql_conda.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\PyQt5\QtSerialPort.pyd
[... cut ...]
LoadLibrary \Device\HarddiskVolume3\Windows\System32\mswsock.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\gnuradio\pdu\pdu_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Library\bin\gnuradio-pdu.dll
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\gnuradio\fft\fft_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume3\Users\rwnst\radioconda\Lib\site-packages\gnuradio\filter\filter_python.cp311-win_amd64.pyd
```

ShoeBro, 2023-12-03:
At least I know how to get it working for now, which is good enough for me. If something else changes or I discover something I will be back here.

Thanks for all the support! If I should close this thread just let me know.

ainst, 2023-12-26:
I am having a very similar problem on a clean install of Windows 11 with no existing Python installation (but one previous non-radioconda version of gnuradio which is now removed).

The first time I run the `dlltracer.trace` command, I have two failed DLLs:
- icudt73.dll
- Qt5Quick3DUtils_conda.dll
but the second time I run it everything is fine.

Interesting `from PyQt5 import Qt` appears twice in my GRC generated Python code. If I delete the first occurrence (above from `gnuradio import qtgui`) then I can run my flowgraph from within ipython with `%run ./test.py`.

If I run the unmodified test.py (as generated by GRC) in ipython, I get the DLL error. From that point on, it is not possible to run, even after modification and `dlltracer.trace `reports about 50% of the DLLs as failed.

There is nothing strange looking in the environment variables.

grmart, 2024-02-10:
I have the same problem with a clean install on Windows 10
radioconda 2024.01.26 (GNU Radio 3.10.9.2, Python 3.11.7)

When I edit the generated python file by removing the first occurrance of "from PyQt5 import Qt" and run it manually in Conda Prompt it works.

ShoeBro, 2024-02-11:
I did a clean install on a new pc and radioconda had no issue, but when I installed a package called openEMS I had the same issue but with openEMS. On my old PC I had installed openEMS first and then radioconda, which gave me this issue with radioconda and openEMS worked just fine.

My solution currently for openEMS was to let python know exactly where the dll's are with this at the start of the file:
```
import os
os.add_dll_directory(r"C:\openEMS")
```

I did not get an opportunity to try this on my old PC with radioconda, but I hope you can use it to maybe get radioconda working. I am also unsure of how radioconda works with the dll's so it may not even be a suitable solution, but I thought it may help!

grmart, 2024-02-11:
I searched a bit and found that it starts working when I manually install "pip install pyqt5-tools" in Conda Prompt and set the environment variable QT_PLUGIN_PATH to "<install path>\radioconda\Library\plugins" (in my case "C:\Users \Martin\radioconda\Library\plugins")

Withou seting QT_PLUGIN_PATH I got another error: "qt.qpa.plugin: Could not load the Qt platform plugin "windows" in "" even though it was found."

I don't understand exactly why it works...

grmart, 2024-02-13:
I edited the generated python file as follows:
```
import sys
import dlltracer

with dlltracer.Trace(out=sys.stdout):
    from PyQt5 import Qt
    from gnuradio import qtgui
    from gnuradio import analog
    from gnuradio import blocks
    from gnuradio import gr
    from gnuradio.filter import firdes
    from gnuradio.fft import window
    import sys
    import signal
    from PyQt5 import Qt
    from argparse import ArgumentParser
    from gnuradio.eng_arg import eng_float, intx
    from gnuradio import eng_notation
    import sip
```
Output:
```
LoadLibrary \Device\HarddiskVolume6\Windows\System32\kernel.appcore.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\Qt.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtXmlPatterns.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Core_conda.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5XmlPatterns_conda.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\ole32.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\combase.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\mpr.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\shell32.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\icuin73.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\userenv.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\pcre2-16.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\icuuc73.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\zstd.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\netapi32.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Network_conda.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\vcruntime140_1.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\crypt32.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\winmm.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\msvcp140_1.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\msvcp140.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\icudt73.dll
Failed \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\icudt73.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\libssl-3-x64.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\libcrypto-3-x64.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\netutils.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\srvcli.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\cryptbase.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\icudt73.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\sip.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtCore.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtNetwork.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtXml.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Xml_conda.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtWinExtras.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5WinExtras_conda.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Widgets_conda.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\dwmapi.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Gui_conda.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Gui_conda.dll
Failed \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\Qt5Gui_conda.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\uxtheme.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\libpng16.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\d3d11.dll
LoadLibrary \Device\HarddiskVolume6\Windows\System32\dxgi.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtGui.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\PyQt5\QtWidgets.pyd
[... 324 lines cut ...]
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\DLLs\_decimal.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\DLLs\_socket.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\DLLs\select.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\matplotlib\backends\_backend_agg.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\gnuradio\blocks\blocks_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\gnuradio-blocks.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\sndfile.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\ogg.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\vorbisenc.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\FLAC.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\mpg123.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\libmp3lame.dll
[... cut ...]
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\gnuradio-pdu.dll
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\gnuradio\fft\fft_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\gnuradio\filter\filter_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Lib\site-packages\gnuradio\analog\analog_python.cp311-win_amd64.pyd
LoadLibrary \Device\HarddiskVolume6\Users\Martin\radioconda\Library\bin\gnuradio-analog.dll
```

ryanvolz, 2024-02-13:
> This is after I removed "C:\Program Files\Elmer 9.0-Release\bin" from PATH and it worked:

Interesting. I see that it tried to load `qwt.dll` and `libgcc_s_seh-1.dll` from there before you removed it, which is likely to cause issues when they are different versions than the ones it is expecting. What did your `PATH` look like before removing it? If the Elmer bit came after radioconda in the path, it shouldn't have found those. But if it came before, well then that explains it.

We might be talking about multiple related issues in this report if people have different things installed that all provide alternative libraries. I don't know if there's a robust way to avoid these interactions when different software modifies and depends on the `PATH`.

ryanvolz, 2024-02-13:
Does setting `CONDA_DLL_SEARCH_MODIFICATION_ENABLE=1` help for anyone? At a prompt, before running anything:
```
set CONDA_DLL_SEARCH_MODIFICATION_ENABLE=1
```

grmart, 2024-02-14:
It's weird because Conda adds its folders at the beginning of the PATH so I don't know why it uses the DLL from the Elmer folder
This is my path in Conda Prompt that causes the DLL error:
```
Path=C:\Users\Martin\radioconda;C:\Users\Martin\radioconda\Library\mingw-w64\bin;C:\Users\Martin\radioconda\Library\usr\bin;C:\Users\Martin\radioconda\Library\bin;C:\Users\Martin\radioconda\Scripts;C:\Users\Martin\radioconda\bin;C:\Users\Martin\radioconda\condabin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files\Microsoft MPI\Bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files\doxygen\bin;C:\Program Files (x86)\Graphviz2.38\bin;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\nodejs;C:\Program Files\Git\cmd;C:\Program Files\TortoiseGit\bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit;C:\Program Files\PuTTY;C:\Program Files (x86)\eID_klient;C:\Users\Martin\AppData\Local\Programs\Python\Python311\Scripts;C:\Users\Martin\AppData\Local\Programs\Python\Python311;C:\Users\Martin\AppData\Local\Microsoft\WindowsApps;C:\Users\Martin\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Martin\AppData\Roaming\npm;C:\Program Files (x86)\mingw-w64\i686-7.3.0-posix-dwarf-rt_v5-rev0\mingw32\bin;C:\Program Files\CMake\bin;.
```

> Does setting CONDA_DLL_SEARCH_MODIFICATION_ENABLE=1 help for anyone? At a prompt, before running anything

It doesn't help in my case.

ryanvolz, 2024-05-21:
I think I now understand what is happening here, at least for the case that I was just able to reproduce. See https://github.com/conda-forge/pyqt-feedstock/issues/138 for a detailed description. But basically, if you have a directory on your `PATH` that contains `Qt5Core.dll` and also a library gnuradio-qtgui needs, like `qwt.dll`, then it will mistakenly try to load that external library instead of the conda-supplied one that it needs. It only happens to prefer that directory because of a bug with PyQt5, which is what I hope to have fixed in conda-forge.

As a temporary workaround, you can navigate to inside the radioconda installation directory and go to `radioconda\Library\bin`. There you can either copy `Qt5Core_conda.dll` and name the copy `Qt5Core.dll`, or just create an empty file named `Qt5Core.dll`. That will short-circuit the PyQt5 bug and make it so the correct libraries are loaded.

I plan to either have `PyQt5` fixed in conda-forge or add a workaround to the `gnuradio-qtgui` package, and then I'll get that fix in the next release of radioconda.

haniyrasul, 2024-05-30:
> Thanks for the report. Evidently some library is failing to load, and we need to figure out what library that is and why. Can you do the following?:
>
> 1. Open the "Conda Prompt" as an Administrator [...]

I have followed the instructions, but it returns an error like this.
[image]

Similarly, when I try to run the Python script of a file, it returns the following error:

> from gnuradio import qtgui
> ModuleNotFoundError: No module named 'gnuradio'

ryanvolz, 2024-05-30:
Fixed for at least one instance in 2024.05.29 release. Please re-open if you still see this same error after upgrading.

closed 2024-05-30 by ryanvolz