radioconda/radioconda-installer #82: issue with UDP-SINK

Lunny1993, 2023-12-14:
1、My gnuradio by radioconda on windows version：
[image]
2、My flowgraph to get and send spectrum data over UDP-SINK：
[image]
[image]
3、In fact, after the flowgraph runs, only a few packets of UDP data can be sent, and then the UDP packets stop sending.
[image]
4、I tried lowering the sample rate or lowering the FFT size etc, but it still didn't work.
5、I noticed the "NOTE" for "UDP-SINK", but how to add the following lines to "/etc/sysctl.conf" on the windows version?
[image]

Can anyone fix it? Ask for advice! Thanks a lot!

ryanvolz, 2023-12-14:
Hi @Lunny1993! I was going to suggest that you post this over at https://github.com/gnuradio/gnuradio where far more eyes would see it and be able to help, but I see you've already done that! I personally have little experience with the UDP Sink, but I would be suspicious of the high packet data size you're trying to use. If you have jumbo frames enabled (and I have no idea how to do this on Windows) then your packet size works, but it doesn't for standard frames. Do you get better results if you change the packet data size back down to the default of 1472? I know that doesn't fit a whole FFT, which I assume is your intention by setting it to `fft_size * 4` where the `4` comes from the 4 bytes per float32 sample.

You might also to well to look into the "ZMQ PUB Sink" or "ZMQ PUSH Sink" as alternatives for sending packetized data to another application. That only works if you can use ZeroMQ on the other end though.

yourarchipelago, 2024-01-04:
I found this issue last year.

ryanvolz, 2024-05-09:
Probably best handled upstream at the GNU Radio issue tracker on GitHub if there are still issues with the UDP Sink on Windows.

closed 2024-05-09 by ryanvolz