radioconda/radioconda-installer #86: Unable to choose editor for Python block on MacOS

techfoamer, 2024-02-06:
Message on console when clicking "Open in Editor" or "Choose Editor" is:
GLib-GIO-WARNING **: 18:00:23.566: No default handler found for content type 'dyn.agq81k3p2su11upputf4gu55s'

Observed this on MacOS 14.3 on both Intel and Apple silicon.

ryanvolz, 2024-02-06:
If you make the following edit to `$CONDA_PREFIX/lib/python3.11/site-packages/gnuradio/grc/gui/Dialogs.py` (replace `if content_type == "*":` with `if Gio.content_type_is_unknown(content_type):`):
```
--- a/grc/gui/Dialogs.py
+++ b/grc/gui/Dialogs.py
@@ -408,8 +408,8 @@ def choose_editor(parent, config):
     Give the option to either choose an editor or use the default.
     """
     content_type = Gio.content_type_from_mime_type("text/x-python")
-    if content_type == "*":
-        # fallback to plain text on Windows if no useful x-python association
+    if Gio.content_type_is_unknown(content_type):
+        # fallback to plain text if no useful x-python association
         content_type = Gio.content_type_from_mime_type("text/plain")
     dialog = Gtk.AppChooserDialog.new_for_content_type(
         parent,
```
does that fix it?

techfoamer, 2024-02-06:
It changes it.  I now have some choices of editors, if I select any of the editor options and click "Open in Editor" the console shows this error:
"Error opening an external editor. Please select a different editor."

ryanvolz, 2024-02-06:
Looks like that's going to require some digging around [here](https://github.com/gnuradio/gnuradio/blob/main/grc/gui/canvas/flowgraph.py#L125) to see why it doesn't work on a Mac. It might be time to open a new issue upstream or raise it in the Matrix chat to see if someone with a Mac can help debug.

ryanvolz, 2024-05-08:
@WTPhD There is a little more discussion over at https://github.com/gnuradio/gnuradio/issues/7115, but still no good solution. I'd be interested in hearing if the change I've proposed in https://github.com/gnuradio/gnuradio/issues/7115#issuecomment-1932586390 helps at all when "Choose Editor" is used (probably also the change in https://github.com/ryanvolz/radioconda/issues/86#issuecomment-1930406202 is needed as well).

ffu, 2025-03-26:
Here are some issues I found. First part is that Apple does not recognize "text/x-python" as a MIME. Apple uses UTI, and if I change to "text/x-python-script", I could get the menu of editors. Second issue is that the AppInfo method get_executable or get_commandline will not return the full executable path of the chosen app. This works for Linux, but not Apple. 

So, one possible fix is to check for MacOS and if so, search the /Application directory to find the correct executable path. The other solution is to do it the "right" way and use AppInfo's launch method. This requires some fairly significant code changes, because GRC currently uses a subprocess to run the app.